n0bel/PiClock #1: maps not displaying in radar

jcwho, 2015-06-11:
I've configured PiClock to only display the time/weather data ( no sensors, no ambient lights, etc ). I have confirmed my google API key is correct but the maps are not displayed in the radar. Any welcome in this simple matter will be appreciated.

n0bel, 2015-06-11:
A few things to try to help debug this...    1) Let the PiClock boot up, then use F4 to exit the clock (its built into the piclock).  Now open a terminal window and type 
...
cd PiClock/Clock
python PyQtPiClock.py Config-Example
...
Does that work?... If so this tells us there's some issue with your specific Config.py

To go further, use F4 again, and your terminal window will come back and you'll see any errors that were generated by python.

Now do it again, with your personal config (leaving off the config argument defaults to Config.py):
...
python PyQtPiClock.py 
...

Does that work?
Then use F4 to close the clock and see if python generated any error messages.

If you could send me your Config.py,  (maybe change all your lattitude/longitude number slightly, and paste bin?, if you do change stuff, be sure it still fails.)  I wish GitHub had a PM function. 

A thought: are you using WiFi or wired?   If WiFi, I've had trouble with WiFi dongles failing to work with large amounts of data.   This was my fix:
...
sudo cp /etc/network/interfaces /etc/network/interfaces.backup
sudo nano /etc/network/interfaces
...
at the bottom of the file add:
...
wireless-power off
Ctrl+x
Y
Enter
sudo reboot
...

n0bel, 2015-06-11:
oh looks like my email is on my profile page: https://github.com/n0bel/ if you want to send something privately, there it is.

jcwho, 2015-06-11:
First, my raspberry pi is headless so i installed PiClock in my Gnu/Linux box and it works perfectly except for the maps in the radar screen. Pressing the space bar switches between the two windows as expected but F4 exits PiClock. The only output is ( in all tests above ) is:

>>>>>
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
<<<<<<

The connection to the Internet is wired and is working fine, after all the weather data is displayed correctly.  I've attached my Config.py as an image, enough to see nothing is really wrong with it.

I read you're planning to modify the project and I'd not mind waiting for a new release. 
Thanks,
j.c.

[image]

n0bel, 2015-06-11:
the attachment didn't work.
just shoot it to my email.
Waiting for a new release... well.. if I can't reproduce the problem, it'll never get fixed.
Apparently, that linux distro doesn't like the .png files I've created, so maybe there's more incompatibilities, that I'm not aware of.    What distro and version are you using?

jcwho, 2015-06-11:
Here is is, a copy of my Config.py with modified coordinates.

  j.c.

n0bel, 2015-06-11:
no attachment came thru.. probably it doesnt work via email, you probably need to do it in github.

jcwho, 2015-06-11:
Dear Mr. Uhlir,

Here is the contents of the file i tried to attach:

>>>
from GoogleMercatorProjection import LatLng

wuprefix = 'http://api.wunderground.com/api/'
wulocation = LatLng(44.76267422986176,-79.99219036102295)
noaastream = 'http://audioplayer.wunderground.com:80/tim273/edina'
background = 'images/clockbackground-kevin.png'
squares1 = 'images/squares1-kevin.png'
squares2 = 'images/squares2-kevin.png'
icons = 'icons-lightblue'
textcolor = '#bef'
clockface = 'images/clockface3.png'
hourhand = 'images/hourhand.png'
minhand = 'images/minhand.png'
sechand = 'images/sechand.png'

radar1 = {
    'center' : LatLng(44.76267422986176,-79.99219036102295),
    'zoom' : 2,
    'markers' : (
        {
        'location' : LatLng(44.76267422986176,-79.99219036102295),
        'color' : 'red',
        'size' : 'small',
        },
        )
    }

radar2 = {
    'center' : LatLng(44.76267422986176,-79.99219036102295),
    'zoom' : 3,
    'markers' : (
        {
        'location' : LatLng(44.76267422986176,-79.99219036102295),
        'color' : 'red',
        'size' : 'small',
        },
        )
    }

radar3 = {
    'center' : LatLng(44.76267422986176,-79.99219036102295),
    'zoom' : 4,
    'markers' : (
        {
        'location' : LatLng(44.76267422986176,-79.99219036102295),
        'color' : 'red',
        'size' : 'small',
        },
        )
    }

radar4 = {
    'center' : LatLng(44.7212951,-93.2008627),
    'zoom' : 5,
    'markers' : (
        {
        'location' : LatLng(44.7212951,-93.2008627),
        'color' : 'red',
        'size' : 'small',
        },
        )
    }
<<<

j.c.

n0bel, 2015-06-11:
Ok.. it works on my Pi and on Windows.   Here's the windows capture..

https://github.com/n0bel/PiClock/blob/master/Documentation/2015-06-11_10-36-44.png

I see you have the zoom factors way out for a big picture... I'm actually surprised that works.. but YAY! it does.   I may change one of my pages for that too.. it looks cool...

So.. Next step would be for me to install on the Linux distro and version you're using.  I'll take some time over the next week to install a VM.   What Distro and Version are you using?

jcwho, 2015-06-11:
Thanks for the attention to this matter.
  I'm running fedora 22; here is the output of "uname -a"

4.0.4-303.fc22.x86_64 #1 SMP Thu May 28 12:37:06 UTC 2015 x86_64 x86_64
x86_64 GNU/Linux

I attached a complete list of packages installed in my host. Hopefully your
e-mail client will not filter it.
Regards,

j.c.

a52dec-0.7.4-19.fc22.x86_64
aalib-libs-1.4.0-0.26.rc5.fc22.x86_64
abattis-cantarell-fonts-0.0.16-2.fc22.noarch
abrt-2.5.1-3.fc22.x86_64
abrt-addon-ccpp-2.5.1-3.fc22.x86_64
abrt-addon-coredump-helper-2.5.1-3.fc22.x86_64
abrt-addon-kerneloops-2.5.1-3.fc22.x86_64
abrt-addon-pstoreoops-2.5.1-3.fc22.x86_64
abrt-addon-python-2.5.1-3.fc22.x86_64
abrt-addon-python3-2.5.1-3.fc22.x86_64
abrt-addon-vmcore-2.5.1-3.fc22.x86_64
abrt-addon-xorg-2.5.1-3.fc22.x86_64
abrt-cli-2.5.1-3.fc22.x86_64
abrt-dbus-2.5.1-3.fc22.x86_64
abrt-desktop-2.5.1-3.fc22.x86_64
abrt-gui-2.5.1-3.fc22.x86_64
abrt-gui-libs-2.5.1-3.fc22.x86_64
abrt-java-connector-1.1.0-4.fc22.x86_64
abrt-libs-2.5.1-3.fc22.x86_64
abrt-plugin-bodhi-2.5.1-3.fc22.x86_64
abrt-python-2.5.1-3.fc22.x86_64
abrt-python3-2.5.1-3.fc22.x86_64
abrt-retrace-client-2.5.1-3.fc22.x86_64
abrt-tui-2.5.1-3.fc22.x86_64
accountsservice-0.6.40-1.fc22.x86_64
accountsservice-libs-0.6.40-1.fc22.x86_64
acl-2.2.52-7.fc22.x86_64
adf-tribun-fonts-1.13-9.fc22.noarch
admesh-libs-0.98.2-1.fc22.x86_64
AdobeReader_enu-9.5.5-1.i486
adobe-release-i386-1.0-1.noarch
adobe-release-x86_64-1.0-1.noarch
adobe-source-code-pro-fonts-1.017-5.fc22.noarch
adobe-source-han-sans-cn-fonts-1.002-1.fc22.noarch
adobe-source-han-sans-twhk-fonts-1.002-1.fc22.noarch
adobe-source-sans-pro-fonts-1.050-4.fc22.noarch
advancecomp-1.19-3.fc22.x86_64
adwaita-cursor-theme-3.16.2.1-1.fc22.noarch
adwaita-gtk2-theme-3.16.2-1.fc22.x86_64
adwaita-icon-theme-3.16.2.1-1.fc22.noarch
agg-2.5-21.fc22.x86_64
aic94xx-firmware-30-6.fc20.noarch
aisleriot-3.16.2-1.fc22.x86_64
akmods-0.5.2-1.fc22.noarch
akonadi-1.13.0-12.fc22.x86_64
alarm-clock-applet-0.3.3-11.3.x86_64
albatross-gtk2-theme-1.7.3-2.fc22.noarch
albatross-gtk3-theme-1.7.3-2.fc22.noarch
albatross-xfwm4-theme-1.7.3-2.fc22.noarch
aldusleaf-crimson-text-fonts-0.8-0.4.20130806.fc22.noarch
allegro-4.4.2-11.fc22.x86_64
alsa-firmware-1.0.29-1.fc22.noarch
alsa-lib-1.0.29-1.fc22.i686
alsa-lib-1.0.29-1.fc22.x86_64
alsa-plugins-pulseaudio-1.0.29-1.fc22.x86_64
alsa-tools-firmware-1.0.29-1.fc22.x86_64
alsa-utils-1.0.29-1.fc22.x86_64
am-utils-6.2.0-2.20140906gitbb13dea6.fc22.x86_64
[... 1824 lines cut ...]
libwps-0.3.1-2.fc22.x86_64
libwsman1-2.4.12-1.fc22.x86_64
libwvstreams-4.6.1-12.fc22.x86_64
libX11-1.6.3-1.fc22.i686
libX11-1.6.3-1.fc22.x86_64
libX11-common-1.6.3-1.fc22.noarch
libX11-devel-1.6.3-1.fc22.i686
libX11-devel-1.6.3-1.fc22.x86_64
libx86-1.1-20.fc22.x86_64
libXau-1.0.8-4.fc22.i686
libXau-1.0.8-4.fc22.x86_64
libXau-devel-1.0.8-4.fc22.x86_64
libXaw-1.0.12-4.fc22.x86_64
libxcb-1.11-5.fc22.i686
libxcb-1.11-5.fc22.x86_64
libxcb-devel-1.11-5.fc22.x86_64
libXcomp-3.5.0.31-1.fc22.x86_64
libXcompext-3.5.0.31-1.fc22.x86_64
libXcomposite-0.4.4-6.fc22.i686
libXcomposite-0.4.4-6.fc22.x86_64
libXcomposite-devel-0.4.4-6.fc22.x86_64
libXcompshad-3.5.0.31-1.fc22.x86_64
libXcursor-1.1.14-4.fc22.i686
libXcursor-1.1.14-4.fc22.x86_64
libXcursor-devel-1.1.14-4.

n0bel, 2015-06-12:
Another user reported: "When asking for a new key it did not activate the static maps part. I had to enable it."

Beyond that, I've just updated the software to remove the google maps api key requirement. 

This should work for low usage. 
do the following, to update your PiClock software and remove the api key.

cd PiClock
git pull
nano Clock/ApiKeys.py

change the google api key to an empty string

googleapi = ''

see if that works

If not, I'll continue to load up FC22 in a VM.

jcwho, 2015-06-12:
Dear Mr Uhlir,

  That did the trick.
  Thank you again for looking into this and have a nice weekend.

  J.C. Oliveira

n0bel, 2015-06-12:
Great!

closed 2015-06-14 by n0bel