n0bel/PiClock #3: Unknown Crash of PyQtPiClock.py

n0bel, 2015-06-15:
As of commit ed7380d7120417bab2a39bca8fdac15dc6f4cafc, PyQtPiClock.py stops working sometimes.  I'm currently investigating why.

n0bel, 2015-06-18:
The crash seems to be related to overlapping calls to get the radar map.  This could have been caused by flipping pages too fast.  Attempt to fix in https://github.com/n0bel/PiClock/commit/149b14c63221416af844c41163e03e3bd36b46ac

n0bel, 2015-06-30:
I had some crashes today.   Segmentation Faults.. I've not had any for days and days, and now 2 in one day...  The last time this happened, Weather Underground was having issues.  It always happens when processing newly downloaded gif files. My suspicion is invalid or malformed gif files.  I'm not sure how to fix this, since it won't be possible to trap Segmentation Faults.

n0bel, 2015-07-11:
I believe I've traced the crashes to corrupt or empty radar image gif files.   I've made a few tweaks that may help prevent crashes.  I've had no crashes for 12 days.

martymoose, 2015-08-03:
working straight for 10 days
Can the Daily rainfall be added (Total for the 24 hours)

n0bel, 2015-08-04:
This one is put to bed i think.

closed 2015-08-04 by n0bel