n0bel/PiClock #4: temp sensor

OKenobi, 2015-06-27:
hello,

Got the temp sensor (365buying DS18B20 Temperature Sensor - Waterproof Digital Thermal Probe Sensor DS18B20) from Amazon.
I just don't know how often the sensor reads the temp. definetely is not real time as I've tested it in boiling water and water with ice. no changes in readings unless I'm restarting the PiClock. I beleive it is a real-time reading form temp sensor but PiClock doesn't give me the feedback in real-time and not sure how often refreshes the reading.

another issue I've noticed is that in the right side of screen where forecast is displayrd for the following days, the 4th box from top displays the wrong day (eg. fri, fri, sat, FRIDAY, sat, sun, mon, tue, wedn) regardless what day of the week it is it looks like is displaying the name of the current day.

thank you.

OKenobi

n0bel, 2015-06-27:
The temperature is update every 10 minutes.

The forecast boxes are designed to do this:  The first 3 blocks on the right show now+3 hours now+6 hours and now+9 hours.   The last 6 blocks show today, tomorrow, next day, etc.

OKenobi, 2015-06-27:
now sky is clear and is all good :)

thank you.

PS: great project, btw. keep up the good work. kudos. :)

closed 2015-06-27 by n0bel