n0bel/PiClock #10: Temperature units

Wavey1978, 2016-01-24:
Hi, Is it possible to change the temp units to degrees Celsius instead of Fahrenheit for those in Europe?

Nielske62, 2016-01-24:
In the config file there is an option to change to metric (metric = 0), if you change this to 1 it will be celcius.

n0bel, 2016-01-24:
thanks for posting the answer Nielske62

closed 2016-01-24 by n0bel

jratha, 2017-09-16:
HI Kevin

Having problems with the inside temp using the DS18B20 Temperature sensor.  My clock is set to metric
[image]

But as you can see, the inside temp is still in Fahrenheit.  

Can you let me know how this can be changed

Thanks

n0bel, 2017-09-16:
There is code to show the inside temperature in metric.   As with the previous posts in this old thread, it is controlled by the config, and that setting is respected throughout the code.   

The real problem is some kind of communication issue between the Pi and the DS18B20.   85C is the temperature returned by the sensor when a conversion has not been triggered.  Since this is all handled in the drive, that leaves some kind of comms issue.  Do you have the pullup?   Is config.txt set correctly?
`dtoverlay=w1-gpio,gpiopin=4`

Another way to prove that it is a comms issue is to hold the sensor in your hand and notice that it doesn't ever change from 85.

You'll see discussions of DS18B20 and the magic number of 85C in the following search:

https://www.google.com/search?q=ds18b20+85c

Hope this helps!