n0bel/PiClock #9: Refresh temerature sensor

Nielske62, 2016-01-24:
I saw in another issue that the refresh rate of the temperature sensor is 10 minutes. Is there a way to make it refresh more often?

PS: Great work with the PiClock, it's amazing!

n0bel, 2016-01-24:
Its doubtful that the thermal mass of the sensors that you would see much of a change over 10 minutes anyway (unless you get some heating by direct sunlight, which defeats the purpose anyway)

But here are the relevant lines of code:

Line 95 of PiClock/Temperature/TempServer.py, the 10 on that line represents the minutes that the sensor(s) are read.   

Line 280 of PiClock/Clock/PyQtPiClock.py, the 10 on that line represents the minutes between display of the temperature.

n0bel, 2016-01-24:
Oh i see you'd have to do something with line 105 too.  That sets a 120 second (2 minute) minimum.

Nielske62, 2016-01-25:
Thank you, works as a charm!

closed 2016-01-25 by Nielske62