n0bel/PiClock #17: Radar Not Working

gshafer, 2016-07-30:
I got all of the features working except radar. On startup I get the following error/messages:

pi@raspberrypi:~/PiClock $ sh startup.sh
numid=1,iface=MIXER,name='PCM Playback Volume'
  ; type=INTEGER,access=rw---R--,values=1,min=-10239,max=400,step=0
  : values=400
  | dBscale-min=-102.39dB,step=0.01dB,mute=1
sudo: Button/gpio-keys: command not found
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "TempServer.py", line 63, in t_http
    server.close()
UnboundLocalError: local variable 'server' referenced before assignment

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "TempServer.py", line 69, in t_udp
    sock.bind(server_address)
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 98] Address already in use

127.0.0.1 - - [30/Jul/2016 17:18:20] "GET /temp HTTP/1.1" 200 -
unclutter: someone created a sub-window to my sub-window! giving up

Any suggestions?

I have tried running the Berlin config with the same issue. I have added the API Keys for both Google and WU.

n0bel, 2016-07-30:
The errors are from the temperature sensor server, which doesn't like being run twice, since its a daemon that stays running until reboot.   Same with unclutter.   The main program PyQtPiClock.py sends its output to log files.

Check the log:
~/PiClock$ ls -ltR Clock/*.log
-rw-r--r-- 1 pi pi 3085 Jul 30 16:22 Clock/PyQtPiClock.5.log
-rw-r--r-- 1 pi pi 3094 Jul 30 16:22 Clock/PyQtPiClock.4.log
-rw-r--r-- 1 pi pi 3094 Jul 30 16:17 Clock/PyQtPiClock.3.log
-rw-r--r-- 1 pi pi 5690 Jul 30 16:13 Clock/PyQtPiClock.2.log
-rw-r--r-- 1 pi pi 3094 Jul 30 15:08 Clock/PyQtPiClock.1.log

Look at the newest dated log.
First check for any other hints of errors.
Second try copying the first "radar url:" and pasting it into a browser, to see what you get.
There are some places in the world that the radar simply doesn't work.  I added the satellite option in those cases.

Also the Google API Key can be '', and with this low volume, its fine.
Also FYI, at this moment, Berlin has no rain, the radar is clear.

gshafer, 2016-07-31:
Kevin,

Thanks for your suggestions.

I rebooted the Raspberry, which fixed the errors.

Looks like the radar was actually working, it was the Google Maps that were not. I put in an API, but it was being rejected. Just left it blank and then maps works fine.

All is working great now. Your project is live and working great in Canada.

Thanks for sharing the project. Great work!

....George

n0bel, 2016-07-31:
Glad its working.. have fun!   Share a picture of it!

closed 2016-07-31 by n0bel