n0bel/PiClock #18: Unable to autostart

sscastor, 2016-07-31:
I've typed in the line as listed for Autostart on reboot however it just goes to the desktop.  When I manually run startup.sh the 1st line is an error that states it is unable to cd to home/pi/PiClock but it eventually loads as it should. I'm not very versed in Linux so very specific instructions are appreciated.

n0bel, 2016-07-31:
There is a delay in startup.sh of 45 seconds.   Thats the evenutally part.

Several things in startup.sh go into daemon mode and then don't like to be started again, those are probably the error messages.

If you added to the end of the file presented by crontab -e, it should look exactly like what follows.   All the lines that start with # are comments and don't count.   The @reboot line is the thing that causes startup.sh to run.

```

# Edit this file to introduce tasks to be run by cron.
#
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
#
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').#
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
#
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
#
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
#
# For more information see the manual pages of crontab(5) and cron(8)
#
# m h  dom mon dow   command
@reboot sh /home/pi/PiClock/startup.sh
```

sscastor, 2016-07-31:
My file looks identical to yours but it is not loading? Should it load or bypass the desktop?

sscastor, 2016-07-31:
In reading the last line about caps I wonder if I did something similar. As I try to manual go thru the directories listed I do not see a home. When I type sudo su and do an ls I see PiClock as the only listing? Again I'm not very well-versed in Linux

n0bel, 2016-07-31:
sudo su is for becoming root... this is not needed to start.. the clock does not run as root, but as the user "pi"

The desktop is not bypassed.  It comes up.. startup.sh waits for 45 seconds before starting the clock, which then overlays the desktop.   The desktop is always there.

Based on what you said, I think maybe you did this part:
```
Get the PiClock software

Log into your Pi, (either on the screen or via ssh) (NOT as root) You'll be in the home directory of the user pi (/home/pi) by default, and this is where we want to be.

git clone https://github.com/n0bel/PiClock.git
......
.....
```
of the instructions as root, rather than just the user pi.

sscastor, 2016-07-31:
I think you're correct. Is that a redo? If so I may just manually start it each time :-)

n0bel, 2016-07-31:
I'd redo it yes.  I'd suggest not to run PiClock from root.

If you choose not to redo it, then ...  try this
```
sudo crontab -e
```
and add the line
```
@reboot sh /root/PiClock/startup.sh
```

Still I'd push you to just start completely over... Pull out that SD card, put it aside and start with a new one.   The full set of instructions only takes about 1/2 hour to do... I just ran thru it on a fresh SD to verify the instructions were correct, since I've done a few updates lately.

sscastor, 2016-07-31:
Thanks for your help and you are correct, it really didn't take that long. I guess doing it right will improve that "not very versed in Linux" issue :-)   Great job, enjoyed working on the project.

closed 2016-07-31 by sscastor