n0bel/PiClock #19: fatal: repository 'https://github.com/n0bel/PiClock.gi/' not found

ObelI, 2016-08-01:
I just follow your guide end:

pi@PiClockKitchen:~ $ git clone https://github.com/n0bel/PiClock.gi
Cloning into 'PiClock.gi'...
Username for 'https://github.com': mr.obeliks@gmail.com
Password for 'https://mr.obeliks@gmail.com@github.com': 
remote: Repository not found.
fatal: repository 'https://github.com/n0bel/PiClock.gi/' not found

haxorthematrix, 2016-08-01:
The git clone line repository is missing the final t:

git clone https://github.com/n0bel/PiClock.git

From the Get PiClock portion here:https://github.com/n0bel/PiClock/blob/master/Documentation/Install-Clock-Only.md

n0bel, 2016-08-02:
Yes, thanks haxorthematrix, that's it.
Obell, just do it again with the full repository name.

ObelI, 2016-08-02:
yep works fine now.

Thank you

n0bel, 2016-08-02:
great!

closed 2016-08-02 by n0bel