n0bel/PiClock #20: Not playing weather report

ScottChapman, 2016-08-02:
Might be an issue with my pi config, but I am not getting audio when using weather from [Worcester MA](http://audioplayer.wunderground.com/CW5897/Auburn.mp3)

n0bel, 2016-08-02:
Audio problems over HDMI can usually be solved by the solutions in this post:

https://www.raspberrypi.org/forums/viewtopic.php?t=5062

ScottChapman, 2016-08-02:
Not running audio over HDMI, just (for now) plugged in speaker to audio port.

My issue *might* be related to the audio URL for my location (which appears to be different than most off the others since it appears to be an MP3).

That's why I posted.

n0bel, 2016-08-02:
Your URL works fine here.

In that case look at this:

https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=40872

Which gives a possible solution, add a line after 21 in startup.sh
sudo amixer cset numid=3 1

sudo amixer cset numid=3 <output>

0=auto
1=headphones
2=hdmi

ScottChapman, 2016-08-03:
Yea, so maybe I am doing something wrong. I tried that line in my startup. I also verified that speaker_test works (which it does). But I am still not getting the forecast to play.

Any other ideas?

n0bel, 2016-08-03:
Does it play manually?
The command is:
```
mpg123 http://audioplayer.wunderground.com/CW5897/Auburn.mp3
```

ScottChapman, 2016-08-03:
Yes.

n0bel, 2016-08-03:
Does it work when you press the F2 on a keyboard connected to the pi?

ScottChapman, 2016-08-03:
No it does not. I've attached my Config.py file
[Config.py.zip](https://github.com/n0bel/PiClock/files/398411/Config.py.zip)

n0bel, 2016-08-03:
There may not be anything revealing, but reboot, do an F2, wait about 30 seconds, to be sure its still failing.   Then use F4 to close the clock.   Then review the log files.   Look at the log file with the latest date/time.  ls -ltr PiClock/Clock/*.log

n0bel, 2016-08-03:
I deleted the previous comment since you have your lat/lon in it.

ScottChapman, 2016-08-03:
OK, I see this in the logs:

Traceback (most recent call last):
  File "PyQtPiClock.py", line 556, in keyPressEvent
    weatherplayer = Popen(["mpg123","-q",  Config.noaastream ])
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

n0bel, 2016-08-03:
somehow you skipped the following in the instructions:

##Get mpg123 (optional to play NOAA weather radio streams)
(you must still be root [super user])
```
apt-get install mpg123
```

At this point just do:

```
sudo apt-get install mpg123
```

ScottChapman, 2016-08-03:
Boom.

Not sure how I missed it. Thanks!

n0bel, 2016-08-03:
great!

closed 2016-08-03 by ScottChapman

n0bel, 2017-06-24:
NOAA weather radio streams have reappeared on the net.  You can find it here: http://noaaweatherradio.org/ They don't put the .mp3 urls where they are easily accessable, so you need to use your browser to "View Page Source" in order to find the proper .mp3 url.