n0bel/PiClock #22: better buttons

ScottChapman, 2016-08-04:
I poked around a bit looking for suitable buttons for my build, and I found [this](https://www.amazon.com/gp/product/B01E43G7I2/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)

What I like about it is that it is a simple membrane button strip I can put on the wall or the bezel of the display.

It occurred to me that you could use these as general navigation buttons (up, down, left, right) by just putting new images on them.

Have you thought about having a more conventional navigation scheme, perhaps where left/right moves between different modes (clock, radar, forecast) and the up/down move between details (analog/digital clock, state/regiom radar, hourly/daily forecast)?

n0bel, 2016-08-04:
Cool buttons.... they'll work fine.

I'm not sure the words "conventional navigation" applies.. since there are only 2 pages currently.

The long term road map includes evolving the config file into 3 pieces.    A layout section (pages, and what 'things' are on the pages and where)   A skin section that drives the look of 'things'.   And then a settings section that give 'things' their settings (location, timezone, language, etc)   'things' I think will eventually be called 'widgets', or hopefully i'll come up with a better name.

At that point your navigation thoughts would work out well.

Also at some point I'll be trying a gesture sensor ( like: https://www.sparkfun.com/products/12787 ), and eliminate the need for any physical buttons.   Unfortunately I've stolen the I2C pins already for the IR remote sensor, so I may have to redo some things.

This is the way I did my buttons.  https://pbs.twimg.com/media/CoohAJZUMAEZR5j.jpg:large

n0bel, 2016-08-04:
Oh and I just received this: https://www.adafruit.com/product/1362

Some tiny wires (wirewrap style) and a bit of copper tape, with a label over it should make nice buttons too.

I'll do a picture when I'm done.

ScottChapman, 2016-08-04:
@n0bel - yea, I was thinking more when there are more pages. They can be grouped by type, and cycle through the types. I think that would be a nice navigation model once you have more pages.

Oh man that gesture sensor is dope!

+1 for replacing the IR sensors with that bad boy!

n0bel, 2016-08-04:
Thanks, I just ordered 2 of the gesture sensors to try out.

ScottChapman, 2016-08-04:
Do you need a shipping address to send the other one? ;-)

n0bel, 2016-08-04:
umm... no cuz i'm currently binge building a bunch for friends and family.

martymoose, 2016-08-04:
Hi have you thought off a lux sensor to change from night/day mode
Just a idea

n0bel, 2016-08-04:
@martymoose  oh sure.. that and 50 other cool things....  I'm happy to review patches and pull requests

gshafer, 2017-05-21:
Can someone share the code changes to make a button like this work? I prototyped a simple button just to get the code working. I am clearly not very much of an expert with python and not clear how to add the code. Here is the simple example code program:

import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)

GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)

while True:
    input_state = GPIO.input(18)
    if input_state == False:
        print('Button Pressed')
        time.sleep(0.2)

I can get the program to work as a stand alone program, but after trying several ideas to add the main program where the keyboard key section is, I have not been able to get it to work. Any help would be greatly appreciated.

n0bel, 2017-05-22:
@gshafer  RPi.GPIO is not inherently event driven as Qt4 would require.   A task tread could be created with the kind of code you suggest, which could then throw events when the button is push (albeit with some debouncing code).    However I chose to use a different method with c code in the Buttons folder and the gpiokeys program.   This program translates gpio button presses to keyboard-like presses.

In startup.sh, you find:
```
sudo Button/gpio-keys 23:KEY_SPACE 24:KEY_F2 25:KEY_UP &
```
Which basically says  gpio23 will be translated to the space key, 24 to the F2, and 25 to the UP key.   Then in PyQtPiClock.py you'll see code using the Qt4 keyboard events to take action based on keys:
```
       if event.key() == Qt.Key_Space:
              nextframe(1)
```
So this would be an approach that would work to map gpio18 to whatever function you were looking for.

closed 2018-12-23 by n0bel