n0bel/PiClock #23: Monitor Support?

dustinarden, 2016-08-04:
I hate posting this in the "issues" as its not one, but more of a question.

I have an old 19inch monitor 4:3 aspect ratio .. would i have to manually modify the screen parameters in the build or is there an option / auto detect for other monitor types?  I'm probably going to replicate this for my new office in the next week or two (just need to buy yet another rPi)

n0bel, 2016-08-04:
Its all automatic.   The program is based on a 16:9, but scales both x&y on everything, including font sizes.

My bedside clock is a good example.   It "looks" 16:9, but it is really 800x600 (4:3).   They apparently size the pixels to make it physically 16x9.

Take a look at this:  http://n0bel.net/v1/index.php/projects/raspberry-pi-clock/44-piclock-gets-config-swapping-feature-day-night-mode-for-bedside-clock    

You can see the clock is distorted (wider than it should be) because the software has no knowledge of the physical aspect ratio.   Instead it was presented with 800x600 (4:3) and computed a square given that info.  

Bottom line is that your 4x3 monitor will work fine.

dustinarden, 2016-08-04:
awesome!  Thanks for the confirmation.

closed 2016-08-04 by dustinarden