n0bel/PiClock #24: Does it matter what version of qt4 I run in Windows if I am using Python 2.7.12?

ghart99999, 2016-08-04:


n0bel, 2016-08-04:
I'm guessing no.  Back in may of 2015 I pulled a copy of http://portablepython.com/wiki/Download/  2.7.6 version.  It includes PyQt4 as shown here http://portablepython.com/wiki/PortablePython2.7.6.1/  .  On the windows side, thats all I know about versions.

ghart99999, 2016-08-04:
Ahh OK. I didn't realize qt4 was included with Python itself. I though it
was a total separate download and install. Thank you.

n0bel, 2016-08-04:
No, PyQt4 is not included in the normal python download.  Its just included in the one from portablepython.com which I happened to use for myself.

ghart99999, 2016-08-04:
OK. Just be safe, I will use the portablepython version you used as well.

n0bel, 2016-08-04:
sounds good

closed 2016-08-12 by n0bel