n0bel/PiClock #27: Basic questions from a beginner... font and time format?

Jsunn, 2016-08-21:
Hello, 

Thanks for the great project, it's my intro into Raspberry Pi.  I have a few questions:
Is there a way to change the font size in the top left corner where the current conditions are displayed? 
There are times when the current condition description text is too long for the space given and it is truncated.

Also, on the right hand side, with the hourly forecast, the word "Thunderstorm" can be truncated depending on the forecast.  

And last question, the time for the sun rise and sunset, can they be displayed in the 12 hour clock format?

Thanks for your help and I appreciate your time, 
-Jsunn

n0bel, 2016-08-21:
There are not configuration files changes that would address these.
However, all of these changes can be made in the code by inspecting the QSS strings that contain font-size:.  You'll see they are all based on a scaling variable so that the font size changes depending on the resolution of the monitor.   As line 755, in part, it says:
```
 font-size: "+str(int(30*xscale))+"px; "
```
So you'd change the 30, up or down to affect the font size.
There are names given to the various widgets, which are somewhat descriptive of what they are for, as in line 754:
```
wxdesc.setObjectName("wxdesc")
```
As far as the time format for sunrise and sunset, thats on lines 175 to 180.  The WUAPI provides this data in 24 hour format, so you'd need to do some code to convert and format it.

Yes there are many forecast and current weather conditions that truncate.  Since I've wanted this to be readable from a distance, I've left the fonts as they are because most of the conditions fit just fine.   Even tho some are truncated, they have always been understandable.   The problem gets even worse trying to anticipate other language condition descriptions.   

Glad to see you making PiClock.   I really enjoy seeing people making and posting their versions.

Jsunn, 2016-08-23:
Thank you for your help.  I will give it a try and post back with my results.  

I was also thinking if there was a way to have the text "scroll" within the frame, both the current condition and the forecast boxes this would also solve the issue of truncated text.  

Thanks, 
-J

closed 2018-12-23 by n0bel