n0bel/PiClock #28: wunderground API usage alert

martymoose, 2016-08-30:
Hi kevin
have you got these before from wu email  "wunderground API usage alert"
Hello,

Your wunderground API key (xxxxxxxxxxxxxxx) exceeded its allotted usage today by making 28 calls within a minute but the limit is 10.

We used one of your raindrops instead of disabling the key for the remainder of the day. You now have 29 remaining raindrops.

We check usage for 24-hour periods based on U.S. Eastern Time.

The plan you are registered for is Stratus - Developer, granting you 500 calls per day with 10 calls per minute limit. To upgrade your plan go to: http://www.wunderground.com/weather/api/d/xxxxxxxxxxxxxxxxxx/edit.html

Details: exceededrate: maxrate: 28 (of 10)
Time: Tue, Aug 30, 2016 00:00 EDT

i have mine and 3 other people i have built clocks for all over 12 months
no updates or changes at all and we all got the emails
any ideas  i think its a WU issue
cheers

BertLindeman, 2016-08-30:
Same here: two messages like this today.

`Your wunderground API key exceeded its allotted usage today by making 11 calls within a minute but the limit is 10.`

Previous I got one (probably correct) message about the daily limit in March.
Since then the rpi2 with 7" touchscreen runs stable.

grep of my latest log:
```
getting current and forecast:Sun Aug 21 09:29:20 2016
getting current and forecast:Sun Aug 21 10:14:28 2016
getting current and forecast:Sun Aug 21 10:59:36 2016
getting current and forecast:Sun Aug 21 11:44:45 2016
getting current and forecast:Sun Aug 21 12:29:53 2016
getting current and forecast:Sun Aug 21 13:15:01 2016
getting current and forecast:Sun Aug 21 14:00:09 2016
getting current and forecast:Sun Aug 21 14:45:17 2016
getting current and forecast:Sun Aug 21 15:30:26 2016
getting current and forecast:Sun Aug 21 16:15:34 2016
getting current and forecast:Sun Aug 21 17:00:42 2016
getting current and forecast:Sun Aug 21 17:45:50 2016
getting current and forecast:Sun Aug 21 18:30:58 2016
getting current and forecast:Sun Aug 21 19:16:06 2016
getting current and forecast:Sun Aug 21 20:01:15 2016
getting current and forecast:Sun Aug 21 20:46:23 2016
getting current and forecast:Sun Aug 21 21:31:31 2016
getting current and forecast:Sun Aug 21 22:16:39 2016
getting current and forecast:Sun Aug 21 23:01:47 2016
getting current and forecast:Sun Aug 21 23:46:55 2016
getting current and forecast:Mon Aug 22 00:32:04 2016
getting current and forecast:Mon Aug 22 01:17:12 2016
getting current and forecast:Mon Aug 22 02:02:20 2016
getting current and forecast:Mon Aug 22 02:47:28 2016
getting current and forecast:Mon Aug 22 03:32:36 2016
getting current and forecast:Mon Aug 22 04:17:45 2016
getting current and forecast:Mon Aug 22 05:02:53 2016
getting current and forecast:Mon Aug 22 05:48:01 2016
getting current and forecast:Mon Aug 22 06:33:09 2016
getting current and forecast:Mon Aug 22 07:18:17 2016
```
Within this time I count 34 api.wunderground calls.

ScottChapman, 2016-08-30:
I recently got the same warnings from wunderground

merrymac, 2016-08-30:
I have the same warnings.  Something has changed with the API, it appears, in relation to the radar images...  HELP please!  Thanks!

merrymac, 2016-08-30:
https://apicommunity.wunderground.com/weatherapi/topics/wundermap-api-does-not-return-any-image

n0bel, 2016-08-31:
Since I pay for a WU subscription because I have lots of PiClocks running for friends and family, I wouldn't see this myself....

However...  Looking at my logs, I see WU started sending blank radar images..
So, merrymac, your link explains the issue exactly.

```
getting radar map radar1:Mon Aug 29 20:19:16 2016
radar map received:radar1:Mon Aug 29 20:19:17 2016
radar map frame count:radar1:0
```
Frame count is how I detect a bad radar animation (frame count = 0).  If this happens it will try again.  However, I placed no limit on the number of retries.   So it would keep trying (ever 5 seconds) until it would shut you down for 24 hours. 

My clock continued to get this type of failure until:
```
getting radar map radar2:Tue Aug 30 18:58:25 2016
radar map received:radar2:Tue Aug 30 18:58:26 2016
radar map frame count:radar2:5
```

I added a retry of 3 to the radar animation retrieval.   I've changed it in such a way that the retry count is not reset until a good animation is retrieved.  So it'll quickly ( 5 seconds) try again 3 times, then go back to the radar_refresh interval (10 min by default) and reset the count when a good animation is received.  This will let it recover from internet bumps quickly, but stop being aggressive after it fails 3 times.  This will prevent it from ever tripping the free API limits, while still maintaining good quality data.

Just as a side note, if you have 2 PiClocks running the same API key, one or the other will probably begin to fail retrieving information because the query limit is 500 per day.  You'll need separate keys per clock, or pay for one of their plans.

Radar refresh is 10 minutes.   Weather conditions and forecast is 30 minutes.
2 radars on the screen at once is 2 * 6 per hour = 12 * 24 = 288 calls per day.
weather is 2 per hour * 24 = 48 calls per day
288 + 48 = 336 per day.

n0bel, 2016-08-31:
Commit a8e37c9ceee6cf88039fe05cb1c061e6d77b3181 adds the retries.

ScottChapman, 2016-08-31:
@n0bel - to upgrade just do another pull? I assume there are no changes so my existing config file should be OK?

n0bel, 2016-08-31:
@ScottChapman yes and yes.

Further details here:  https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#updating-to-newerupdated-versions

merrymac, 2016-08-31:
Hi Kevin,

Since I have my own tweaked version of the clock running, I manually made your changes in Commit a8e37c9 and am currently testing -- but looks great so far.  Thanks for your quick response, even though it isn't really an issue with your code.  :)

closed 2016-09-06 by n0bel