n0bel/PiClock #29: A PiClock of a different flavor?

n0bel, 2016-09-03:
[image]

I subscribe to http://www.hackerboxes.com Every month they send me a box of "stuff". My main reason for doing this is because I'm exposed to some random parts, some of which I've not seen before, and may find useful.

This month's Hackerbox included an Orange Pi Lite. They included instructions and parts for an lcd display and a push button thingy, and other things to get you started.

Well of course I didn't bother with that, and simply went straight to my PiClock. After burning an SD card (with armbian), a few minutes fussing and searching how to connect to my WiFi, and then a few more minutes of apt-get's the PiClock main program was up and running.

Question? Should I target the Orange Pi series of boards as an alternative PiClock platform? Orange Pi Lite (Wifi): $12USD+shipping from AliExpress ($20USD Amazon). Orange Pi One (Ethernet) $10USD+shipping from AliExpress ($17USD Amazon).

dankolbrs, 2016-09-06:
Is there much involved in the OrangePi to specifically target it? 

Maybe just an install script that installs the required packages whether it is on armbian or raspbian could be the way to go. That could ask for lat/lon and API keys during setup as well to give a basic config.

n0bel, 2016-09-06:
I've been thinking about a scripted install.

There are so many optional pieces, I'm not sure it would work for everyone... But maybe if it was billed as only for a basic no option install it would be useful.   I'll wait for more feedback on this.

The Orange Pi appears to need a slightly higher level of expertise than a Raspberry Pi.   The manufacturer gives no support, but the Armbian site does provide pretty good support.   Since Armbian Jessie Desktop is pretty close to Raspbian Jessie, the process for getting the Clock itself up and running is simply a matter of cherry picking the install instructions.    After I solved some basic Orange Pi issues, I had the clock up and going in about 10 minutes.

Set your timezone 
```
sudo dpkg-reconfigure tzdata
```

Get Qt4
```
sudo apt-get python-qt4
```

Setup your Api keys and Config in PiClock/Clock (refer to original docs)

Run it...
```
cd PiClock/Clock
python PyQtPiClock.py
```

closed 2018-12-23 by n0bel