n0bel/PiClock #30: Troubleshooting IR

skifast42, 2016-09-04:
Hi - love the project!  I have the clock up and running just fine and am trying to diagnose an issue with the IR.  

I'm using the suggested key fob and I've confirmed it's set for the right config with the IRCodes perl script (getting the KEY_UP, KEY_DOWN, etc msgs on the screen).  The IR receiver hardware is wired as suggested using Pin 3, but I'm getting no response from the clock from the key fob presses.  Straight keystroke entry works fine (F2, space bar).

I'm not using any of the other optional hardware - no GPIO buttons, temp sensor or LED light strip - just the IR.  Any suggestions on how to troubleshoot why the IR works fine on the perl script but not the clock (besides triple-checking the config file updates)?

n0bel, 2016-09-05:
It appears that the newer versions of lirc do not default to the right device and/or modules.   It could also be a difference in the distributed versions of /etc/lirc/hardware.conf.

So I've updated the instructions for /etc/lirc/hardware.conf thus:

Be sure the DEVICE line appears as follows
```
DEVICE="/dev/lirc0"
```

Be sure the MODULES line appears as follows
```
MODULES="lirc_rpi"
```

I've confirmed this works with older and newer OS and lirc versions on all the Pis

skifast42, 2016-09-05:
Fantastic - that's the fix.  For posterity, I'm on a Pi3.  Thanks again!

n0bel, 2016-09-05:
Great!

closed 2016-09-05 by n0bel