n0bel/PiClock #31: Time Zone

posi211, 2016-09-05:
This is good looking clock/project.  I wanted to build something like this for years.  

I run into a bit of an issue with the config file.  I can get the maps to show up on the lower left, but I have the wrong time and nothing in the upper left weather box.    

In the config file it looks like the only setting is the primary_coordinates and everything should be connected this that?  What I'm I missing?

n0bel, 2016-09-05:
If you're speaking of the main clock time, this is set up in the instructions https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#first-boot-and-configure, as part of the setup in the Internationalization tab.

If you're speaking of the times in the radar image, thes times are part of the radar image that comes from weather underground.   They choose the timezone based on (if i remember correctly) the western edge of the image.   If you live near the edge of a timezone, or have a small zoom factor (bigger area), the problem will occur.   There is no way that I know of to fix this.  The times can be turned off by making changes to the radar url based on the api documentation.

If nothing shows up on the current conditions area (top left box) I'm not sure what that problem could be.

Perhaps examining the log file(s) will point to an issue:  PiClock/Clock/PyQtPiClock.*.log

posi211, 2016-09-05:
I just rebuilt the install and it works just as it should.  Now I want to change it.  
Need to add a news rss feed.

n0bel, 2016-09-05:
Cool! I'll consider any pull requests.

closed 2018-12-23 by n0bel