n0bel/PiClock #32: Analog to Digital Clock Display

gjohanns, 2016-09-06:
I love the work you have done. I am starting to use this in my office. I have one question: as a non-programmer, is there a way for me to make the analog clock into a digital display? I am trying to look up code examples to make the change on my own, but I'm not having very much luck. Thank you!

n0bel, 2016-09-06:
As a non-programmer, probably not.   I'll work out a digital clock option soon.

n0bel, 2016-09-10:
I've added a digital display option.  Pull updates, review Config-Example.py and modify your config to enable it.   It works in commits after https://github.com/n0bel/PiClock/commit/84ce11268bbe08c4127e7ecb6f1bde93ad41d991

blboyd, 2016-09-10:
That is awesome... I have a clock running at work and at home.

I am Director of Information Technology for a private Christian school and I have students constantly coming through my office to look at my PiClock.

ScottChapman, 2016-09-10:
Images?

n0bel, 2016-09-10:
http://n0bel.net/v1/index.php/projects/raspberry-pi-clock/46-piclock-goes-digital

https://github.com/n0bel/PiClock/blob/master/Documentation/Digital%20Clock%20v1.jpg

https://github.com/n0bel/PiClock/blob/master/Documentation/Digital%20Clock%20v2.jpg

n0bel, 2016-09-10:
@blboyd Great!  Share the pics somewhere!

martymoose, 2016-09-10:
Looks great. Thanks

blboyd, 2016-09-12:
[image]
[image]
[image]

Here's a couple snapshots of my installation. I have it swapping configs every hour via crontab.

One feature that would be nice is to make it easier to customize the background and color scheme. Even if you simply provide your source photoshop/illustrator files so we can alter the colors.

@n0bel  - This is an awesome project! I have actually ordered LCD driver boards to power the LCD screen that I ripped out of some old laptops. We're going to use them here at our school.

n0bel, 2016-09-13:
YAY Great pics and customizing... 

Eventually I'll do a layout, theme and skinning system.   It'll take a bunch of code refactoring, but it'll get done, or maybe someone will pitch in to help.   My starting point will be to refactor to separate out the code for current, forecast, radar and clock into separate modules.  Each module will then deal with its own graphic decorations.  Then a layout definition that lets you arrange the modules as you'd like.   The modules could consult skin image definitions, or expect tintable graphics.

I use GIMP not Photoshop, but I didn't bother saving the .xcf files.   I created the initial light blue set of things, and then simply did color rotations and brightness/contrast changes to change it for subsequent styles.     The weather images were downloaded from some royalty free website.   I then ran them all thru a filter neon filter in gimp.   I looked but I don't see where I saved the original icons.   The clock face and hands was created in a similar way.   I created all these things over 2 years ago.   I guess I never figured I'd need the base images anymore.    Everything I have is in the img and icons folders.   If I find some additional files I'll put them up.

Knackster, 2016-09-13:
Good morning Kevin,
I'm glad to see that you are still developing this as I think it's a pretty cool idea. So cool that I just bought another 24" TV just to display PiClock. The reason for the TV is that the USB ports are powered (for the Raspi) and the HDMI input.  This is the TV I bought last week at only $117.00 US https://goo.gl/ajMvqh . Open box item at Amazon.

n0bel, 2016-09-13:
Cool... great stuff... altho your link didnt work, it took me to my amazon orders, not yours. I have a bunch of PiClocks running for friends and family.   I had to make more and I bought 3 of these https://www.walmart.com/ip/Element-ELEFW195-19-720p-60Hz-Class-LED-HDTV/38661262  Refurbished for $69.00 each.   They also have USB with enough current to drive the Pi, +HDMI and the form factor was such that the Pi tucks in nicely behind.

[image]
[image]
[image]

ScottChapman, 2016-09-13:
I did sort of the same thing. Bought an inexpensive TV/Monitor from Costco. Because it was also a TV it had speakers, and it came with it's own wall mount. All for right around $120. Plus there is a nifty little panel on the bottom where I mounted by 4-button switch to control it. Super happy.

[image]

n0bel, 2016-09-13:
Wonderful pics!!

n0bel, 2016-09-17:
Digital clock option implemented https://github.com/n0bel/PiClock/commit/84ce11268bbe08c4127e7ecb6f1bde93ad41d991

n0bel, 2016-10-01:
https://github.com/blboyd Said:

One feature that would be nice is to make it easier to customize the background and color scheme. Even if you simply provide your source photoshop/illustrator files so we can alter the colors.

And I said that I couldn't find my gimp source files.....   Well this is embarrassing.. They are all right there in the gimp folder in PiClock.  

https://github.com/n0bel/PiClock/tree/master/gimp

sting3rmj, 2016-11-02:
I love this project. I just need to figure out how to change the fonts for the digital clock.
[image]

pigulka, 2016-11-02:
Thank n0bel. He did a great job.
Is it possible to display the time in a digital format 24?
greetings Piotr.

n0bel, 2016-11-02:
Pretty much every setStyleSheet on all the widgets use "font-family:sans-serif;"   Thats what you'd change.    clockface.setStyleSheet near line 893.  I've not tried to add fonts to the Pi, so I can't give you direct instructions how to do it.  But once you've accomplished adding fonts, then you'd reference it with font-family in the style sheet.

n0bel, 2016-11-02:
Just look and modify your Config.py
The format string, "digitalformat" on line 29 of your Config.py

The format string specification is here: https://docs.python.org/2/library/time.html#time.strftime

To be 24 hour you should be able to just change the %I to %H and remove the %p

"{0:%H:%M\n%S}"

closed 2016-11-02 by n0bel

pigulka, 2016-11-02:
THanks   :)  it works !

ronstar55, 2018-02-19:
Kevin - Thanks for developing and sharing PiClock. Your instructions are great - I had this installed and running very quickly.  Now I just need to fine tune to meet my tastes.

My first hurdle is to find out how to improved the screen resolution. The map is not as crisp as a Google maps map. There's probably some info in the code somewhere that will need to be changed to work for my monitor. I'll track this down or send another comment if I get stuck.

Thanks again! This app will look great running on a separate monitor in my office.