n0bel/PiClock #33: Design question about the daily forcasts

BertLindeman, 2016-09-11:
The is not a problem, just a question.
The right side of the panel is populated with 9 forecasts.
Three intervals of 3 hours followed by six, each of a complete day.
Looking at your example [https://raw.githubusercontent.com/n0bel/PiClock/master/Pictures/20150307_222711.jpg] 
Current is Saturday 22:37 (or such)
Forecasts of Sunday 01:00 05:00 and 08:00 (? strange intervals ?)

And then the forecast I question:  the complete day Saturday. This one seems rather useless for me as Saturday has almost passed, even if it's earlier on the day. We cover 9 hours in intervals.
For myself I changed to skip today's forecast in ```PiClock/Clock/PyQtPiClock.py```
```
    for i in range(3,9):
        f = wxdata['forecast']['simpleforecast']['forecastday'][i-3]
```
into
```
    for i in range(3,9):
        # /BL 2016-02-24 drop the forcast of today as we have the 3 3-hour] forecasts
        f = wxdata['forecast']['simpleforecast']['forecastday'][i+1-3]
```
I did not create a pull request, I could ***try*** to create one if you agree.

The PyClock is ***so much*** FUN!

n0bel, 2016-09-11:
The anomaly you see has to do with the simplistic way that I chose to pick out data from the hourly forecast.   It makes an assumption as to which array element(s) contain what data.   The WUAPI isn't always consistent.   

As to  the i+1-3 thing...  My preference in the morning (like between 1am and 5am [yes i'm still awake looking at the clock at those times] ) I'd like to see the current day's forecast too, since most of the day is still ahead.   I might be lulled into thinking there will be no rain upcoming simply because it skipped to the next day, and the rain will come during the daylight hours.   This is why it doesn't skip.  

So the proper approach to fix both issues should be for the logic to look at the dates/times of the array elements, and make proper decisions.  For the hourly anomaly you saw, since it would look at the times and show the appropriate data, regardless of which array element it resides in.   For the daily issue I pointed out, the logic would skip a day or not, depending what the current time is.

BertLindeman, 2016-09-11:
:smiley: Thanks for the explanation.
It was just a question.
So closing as I have a valid response.

closed 2016-09-11 by BertLindeman