n0bel/PiClock #34: windowed mode? also getting an error

dfauci, 2016-09-16:
Hi, 

was wondering if it was possible to run this in a window that i can resize? this does a lot of great things but its not quite everything i am looking to use my Pi for, i would like to add a calendar and maybe one other item to my screen to display at all times, thus i would like to be able to resize this application.

secondly, when i try to start PiClock, i am getting this message, and PiClock never launches: "Warning: Missing charsets in String to FontSet conversion" (i have followed and repeated every step of this install guide to ensure i didnt miss any steps. updated to latest rasbian jesse, on an RPi2 Model B)

n0bel, 2016-09-17:
No it doesn't run in windowed mode at all.  I'll think about how best to do this because resizing is not built into the system.   It might be easier to let the config file set the size and position.

Missing fonts shouldn't be a problem.  Are you using a language other than english?  Its possible that the character fonts that come with the desktop gui on the pi don't support some character or another.

JoeSpencer, 2016-09-24:
Im having the same problem with the font warning

n0bel, 2016-09-24:
Missing fonts shouldn't be a problem. Are you using a language other than english? Its possible that the character fonts that come with the desktop gui on the pi don't support some character or another.   Xwindows will substitute a close font.

n0bel, 2016-09-24:
It could be a locale issue.   Try a different one -- this part of the instructions: https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#first-boot-and-configure

Other than that, can you copy the entire error message?

JoeSpencer, 2016-09-25:
Updated the location info as spec'd
here is a screenshot:
[image]

n0bel, 2016-09-25:
This is a long shot.. but try
```
export LC_ALL=C
sh startup.sh -n
```
Also look at the log files  PiClock/Clock/PyQtPiClock*.log   They may tell more about what is going wrong.

n0bel, 2016-09-25:
I've made some changes to startup.sh and the Install instructions for the initial run.  

I've added -s for logging to the screen instead of logging to the log files.   This should make debugging the initial run easier.   As the instructions have been updated, you should do the initial test run as:
```
cd PiClock
sh startup.sh -n -s
```
Now all logging information goes to the screen, so you can see why the PiClock doesn't start.

Most likely any issues are problems in Config.py or ApiKeys.py

To get the update use the instructions from here: https://github.com/n0bel/PiClock/blob/master/Documentation/Install.md#updating-to-newerupdated-versions

JoeSpencer, 2016-09-25:
Wait....are you from Minneapolis?

n0bel, 2016-09-25:
yes

JoeSpencer, 2016-09-25:
Ha, me too. Thanks for the updates and help, I got it up and running. Its pretty sweet!

closed 2018-12-23 by n0bel