n0bel/PiClock #35: Help Needed for remote control

martymoose, 2016-09-20:
hi kevin or anyone else can help  BUT ONLY If you have time. Its no biggy
i have 4 piclocks out there that want a remote to change to radar only and back again on each press
and would like to use these that i have in draw they are very small  and cost $1
[http://www.ebay.com.au/itm/ML-L3-IR-Wireless-Remote-Control-for-Nikon-D5000-D5100-D7000-D3000-D90-D80-OP-/282046461983?hash=item41ab47781f:g:UOQAAOSwoBtW4Rmv]
i have tried for 6 hours to try to get to work but i am lost 
is there a easy way to make the button change the radar screen  beaten me

heres the lirc.conf for nikon ml-l3 remote
http://lirc.sourceforge.net/remotes/nikon/ML-L3
#

begin remote

  name   Nikon2
  flags RAW_CODES|CONST_LENGTH
  eps            30
  aeps          100

  ptrail          0
  repeat     0     0
  gap    99594

      begin raw_codes

          name shutter
             2069   27761     483    1464     484    3413
              565

          name shutter2
             2189   27719     628    1396     625    3405
              553

          name shutter3
             2192   27651     578    1548     560    3461
              486

          name shutter4
             2112   27710     559    1559     510    3460
              485

          name shutter5
             2126   27765     525    1448     572    3389
              626

          name shutter6
             2125   27774     483    1490     484    3462
              537

          name shutter7
             2155   27590     557    1391     602    3509
              635

          name shutter8
             2368   27575     607    1365     628    3355
              622

          name shutter9
             2089   27780     501    1474     500    3445
              567

          name shutter10
             2124   27768     540    1430     589    3387
              618

          name shutter11
             2053   27851     528    1488     484    3461
              488

          name shutter12
             2095   27924     540    1478     488    3413
              484

          name shutter13
             1971   27813     489    1459     488    3407
              512

          name shutter14
             2138   27748     534    1493     530    3534
              464

          name shutter15
             2148   27838     483    1462     484    3414
              529

          name shutter16
             2120   27741     567    1434     515    3541
              484

          name shutter17
             2109   27869     612    1409     545    3407
              490

          name shutter18
             1980   27929     484    1488     485    3449
              484

          name shutter19
             2116   27736     537    1434     628    3482
              463

          name shutter20
             2069   27687     484    1466     493    3572
              583

      end raw_codes

end remote

n0bel, 2016-09-21:
I have 2 of those remotes.   I'll play with it and see what I can do...

However, it seems to me that you should be able to do something like 

name KEY_SPACE

instead of 

name shutter20

but do it on all the occurances of name.

Then put that into /etc/lirc/lircd.conf

Be sure the /etc/lirc/hardware.conf is setup according to the install instructions.

martymoose, 2016-09-22:
mate works a treat
thanks  here the nikkon ml-l3 conf

#
# this config file was automatically generated using lirc-0.8.0(serial)
# on Mon Jul 3 17:45:48 2006
#
# contributed by blux bernd[ AT ] berndlux.de http://berndlux.de
#
# brand: Nikon model no. of remote control: ML-L3 for D50/D70/D70s DSLR
# still Camera devices being controlled by this remote: Nikon D70s DSLR
# still Camera I read out the one Button 20 times, for the case that
# some codes will not work
#
#
begin remote
  name Nikon2
  flags RAW_CODES|CONST_LENGTH
  eps 30
  aeps 100
  ptrail 0
  repeat 0 0
  gap 99594
      begin raw_codes
          name KEY_SPACE
             2069 27761 483 1464 484 3413
              565
          name KEY_SPACE
             2189 27719 628 1396 625 3405
              553
          name KEY_SPACE
             2192 27651 578 1548 560 3461
              486
          name KEY_SPACE
             2112 27710 559 1559 510 3460
              485
          name KEY_SPACE
             2126 27765 525 1448 572 3389
              626
          name KEY_SPACE
             2125 27774 483 1490 484 3462
              537
          name KEY_SPACE
             2155 27590 557 1391 602 3509
              635
          name KEY_SPACE
             2368 27575 607 1365 628 3355
              622
          name KEY_SPACE
             2089 27780 501 1474 500 3445
              567
          name KEY_SPACE
             2124 27768 540 1430 589 3387
              618
          name KEY_SPACE
             2053 27851 528 1488 484 3461
              488
          name KEY_SPACE
             2095 27924 540 1478 488 3413
              484
          name KEY_SPACE
             1971 27813 489 1459 488 3407
              512
          name KEY_SPACE
             2138 27748 534 1493 530 3534
              464
          name KEY_SPACE
             2148 27838 483 1462 484 3414
              529
          name KEY_SPACE
             2120 27741 567 1434 515 3541
              484
          name KEY_SPACE
             2109 27869 612 1409 545 3407
              490
          name KEY_SPACE
             1980 27929 484 1488 485 3449
              484
          name KEY_SPACE
             2116 27736 537 1434 628 3482
              463
          name KEY_SPACE
             2069 27687 484 1466 493 3572
              583
      end raw_codes
end remote

-- 

*Martin Thompson*
*Po Box 1415*
*AYR   QLD    4807*
*M. 0466 005866*
*E.  **marty@martymandy.com
<https://mail.google.com/mail/u/0/?view=cm&fs=1&tf=1&to=marty@martymandy.com>*
*W. **martymandy.com <http://martymandy.com/>*
Check the weather at my place martymandy.com

Supplying online weather stations

closed 2016-09-23 by n0bel