n0bel/PiClock #37: Clock Face Oblong

gshafer, 2016-09-28:
I finally got around to firing up a 7" display I bought on eBay. For some reason the analog screen face is oblong, not round. As I remember it worked fine when I used my HDMI TV as a monitor. I suspect your software if having trouble figuring out the screen size from this cheapie monitor
. The Aspect ratio is 16x9. Is there a config setting to set this?

[image]

n0bel, 2016-09-28:
It is the display itself that is causing the problem..   I have the same issue with my bedside clock.

[image]

Another similar display does not exhibit the problem.

Some 16x9 small LCD displays do not have a 16:9 pixel resolution, but instead their pixels are 4:3.   My bedside clock has 800x600 which is 4:3, but then the display shows it in 16x9, which stretches in the horizontal direction.    Your display may have a mode setting that sets 4:3 or 16:9, but what that will do is leave blank space on the left and right of the display.

My LCD that works fine is 1024x576, which is 16:9.  On that display no distortion occurs.

The PiClock software uses the screen resolution (pixels) to determine scaling.  So it sees 800x600 and scales according to that.  Since the 800x600 is spread across a 16:9 aspect ration instead of 4:3, in effect the pixels are not square.   

In order to fix this I'd have to introduce a distortion in the scaling part of the program, that squeezes the clock by some factor.  For the background circle this isn't really a problem.   For the clock hands, I'm simply doing rotations, which is a simple calculation.  I think it would be a bit harder to shrink and grow the hands depending on their rotation, in order to pre-distort them.

Here's a picture of a different LCD, the same exact physical size as my bedside clock version, but with the 1024x576 resolution (real 16:9), as you can see, it works fine.

[image]

When I ran into the problem with the bedside clock, I just decided to live with it.  Someday I'll replace it with a new LCD at the right resolution.

gshafer, 2016-09-28:
Thanks for the quick reply and thorough explanation.  I just keep causing myself grief and extra work by ordering these cheap displays from China. There always seems to be a problem! I have ordered a new display with the proper aspec ratio and that should solve my issue. My wife is an atist and she is going to paint a picture and I am going to mount the display on the canvas and hide all the electronics behind it. I will send you a pic when it is completed.

closed 2016-09-28 by gshafer

n0bel, 2016-09-28:
Sounds like a great project!  I'll look forward to seeing it.

There are 2 things that should happen with a 1024x576 display.   1) Not distorted 2) Better looking fonts.  On my bedside clock I had to do a bold attribute on everything.

gshafer, 2017-06-04:
I finally got around to finishing the painting integration. I also added Amazon Alexa integration to allow me to easily change screens. I am still planning on adding a gesture sensor so  I can change screens with a wave of the hand.  Thanks for sharing such a great project.

Here is quick video of the project

https://youtu.be/zyORXt4QIc0

dankolbrs, 2017-06-07:
That's an interesting looking billboard!

Reading through this the oblong clock face issue seemed familiar, and thought it was resolved in another thread. Which it was in #39, adding it in here in case someone stumbles across this thread looking for an answer.