n0bel/PiClock #38: WIP - Seperate API calls

dankolbrs, 2016-09-30:
Hey @n0bel ,

I've been playing around with the code, and working on separating out the API calls to a library function. I've got a couple initial plans here:

1. As little _large_ changes to the existing PyQtPiClock.py as possible on the first iteration. I'm going for creating a minimal json object that can be called from PiClock with all the information it's looking for. 
2. Be able to implement DarkSky (the new name for forecast.io) to take over some of the weather data to stretch API calls to those of us that are too cheap to pay for API access. This would probably be better off using subclasses/inheritance rather than the way I'm doing it right now, but hey, it'd be an easier refactor later than it is currently. Right now I'm thinking basically taking the darksky return and translating it to return the same keys as wunderground (this will still require wunderground for radar).
3. Get some unit testing going - I'm pretty new to formal unit testing, but it's definitely worth the learning curve and I'm kind of disappointed I didn't make the effort to learn it previously.
4. **Kind of unrelated**, I've also got ideas of changing this over to using logger.

Currently this is pretty rough, and on my ["seperate_library"](https://github.com/dankolbrs/PiClock/tree/seperate_library) branch. You'll see the majority of the differences in [`Clock/lib`](https://github.com/dankolbrs/PiClock/tree/seperate_library/Clock/lib) and [`Clock/tests`](https://github.com/dankolbrs/PiClock/tree/seperate_library/Clock/tests). I've just been working on the library, and haven't changed PyQtPiClock.py to pull from this library yet.

I'd be happy to get your feedback or input if you'd rather see it going a different direction and can't see yourself merging a PR with this. If you hate it, that's cool too, thanks for putting your project up on Github! :)

-Dan

n0bel, 2016-10-02:
I'm happy to see someone working on this.   I like your idea, as a step toward multiple weather source options.  Making an API for weather, and therefore requiring a standardized data-set for weather information, is not exactly the way I see proceeding forward.

My vision is to make plugins (granted this is another name for libraries), which are given parameters.   The parameters specify the frame number (parent QFrame widget), the size, and other setting relevant to the task.  Further I would separate weather into different components, Current Conditions, Forecast, Radar, Satellite.    Each component would be responsible for its layout given the parameters.  I'd also envision a skinning/theme system, where components can be skinned directly, or tinted based on a themeing system.  Somehow leveraging the QSS styling system of QT4 would be ideal.  In the case of weather plugin(s) I'm sure there would be a common module for getting api data, to ensure that rate limits are not exceeded.  

Why didn't I do this in the first place?  PiClock is my *first* python project, as mentioned here:  http://n0bel.net/v1/index.php/projects/raspberry-pi-clock/30-goodbye-piclock-hello-pyqtpyclock

There are so many things I'd approach differently now, not limited to better naming conventions, and a much more modular and class based style.

As I'm sure you can see, the route I'd like to go involves a major refactor of the whole PiClock structure.  I've debated in my mind back and forth whether to start over with PiClock2, or tag this as Release 1.0, and start on the road of 2.0, with a final merge and release to 2.0 after the work is done.

I'd love to hear your reaction to this, and if you'd like to pursue helping PiClock evolve to this vision.

/K

dankolbrs, 2016-10-03:
That's pretty much what I expected!

I did see you had a PiClock2 repo and knew this was the intention. I've been poking around with this, and while it's rough, I mostly threw this up to point out to you someone is interested and working on it. Life things may be getting in the way over the next few weeks, so I don't think I'll progress on this real far until I get some more free time.

What I've been working on is pretty much just a step in the process, and breaking out the weather into a separate plugin is just a minor first step. There's really a couple of limitations from these APIs, such as the DarkSky API doesn't do radar, so that will still need to go through Wunderground. Also, the API returns are wildly different. There will have to be a step in any plugin that returns a standardized format, so it can be parsed by the PyQtClock regardless of the weather source. I started with the Wunderground API first, just to get the minimal dict object required from the API. Translating darksky or some other weather API plugin to return that same minimal dict shouldn't be too much of an issue and is just a start in plugins. I understand what you're thinking for plugins to handle layout and other things, and while I have no idea off the top of my head how to make that work, figuring it out sounds like a good time.

Regardless, to make the steps you want to take, this is going to require some breaking out and modularization. Whether that is starting from scratch, or iterating over the existing working project, I don't think it makes too much of a difference.

Most of my code isn't on public github, but if it was you'd see this is more interesting and fun than any project I created from scratch. I'd be happy to contribute to this where I can to help you get this where you want it to go.

dankolbrs, 2016-12-20:
Just a quick note on this. I still intend in progressing further on this but a move, new job, and the occasional other shiny thing have distracted me away from sitting down and putting in more time to this.

Maybe now that I'm in a place where PiClock may report temperatures as negative, and radar images contain blue, I'll find more time to dedicate in the next few months.

dankolbrs, 2017-02-21:
Finally admitting I'm just not going to get any further on this, and closing. Time/effort would be better focused on #54 rather then the direction/abstraction I was going for here.

closed 2017-02-21 by dankolbrs

n0bel, 2017-02-21:
@dankolbrs Thanks for the effort on this.   I'd really appreciate some feedback on #54 from you.