n0bel/PiClock #41: Edit properties of radar

Nielske62, 2016-10-02:
I live in The Netherlands with "great" weather, which means a lot of rain. I've noticed that whenever the radar displays these rainshowers the rate of the elapsed time is very high. Is there a way to change this rate? It would also be nice if you could be able to change the color from green to blue. 

Thanks in advance!

n0bel, 2016-10-02:
New radar .gif files are downloaded from weather underground every 10 minutes, regardless of conditions.  The software is entirely dependent on what they provide.   I have heard of various experiences with radar results in different countries being better or worse, even to the point where some countries/cities have no radar images available.  Until PiClock incorporates more weather provider options, there may not be much that can be done.   You may want to check the weather underground api https://www.wunderground.com/weather/api/d/docs?d=layers/radar to see if there are options you could exploit to get a better experience.

Nielske62, 2016-10-02:
Thank you for the response. It seems that you are unable to modify the .gif except for the parameters given on https://www.wunderground.com/weather/api/d/docs?d=layers/radar.

As to the rate of the elapsed time I was able to do something to make it smoother: I've changed a parameter on line 396 in [PyQtPiClock.py ](https://github.com/n0bel/PiClock/blob/master/Clock/PyQtPiClock.py)to `num=8`. This way the radar animation has 3 extra frames (the maximum number of frames is 15 but above 8 the rainshowers dissapear). Because of those 3 frames it makes the animation somewhat smoother to view.

Thanks for your help!

closed 2016-10-02 by Nielske62