n0bel/PiClock #43: Radar not working

GeorgeJirka, 2016-10-29:
Hello.
I try run your perfect project on my CubieBoard and I have got a problem. Radar not working. Can you help me please?
Here is my log file output. Weather number is remove.

-`None
google map base url: http://maps.googleapis.com/maps/api/staticmap?center=49.2381322,17.6717151&zoom=7&size=266x312&maptype=hybrid&markers=color:red|size:small|49.2381322,17.6717151
radar url: http://api.wunderground.com/api/'remove number'/animatedradar/lang:CZ/image.gif?maxlat=50.3444941052&maxlon=19.1328967406&minlat=48.1064200968&minlon=16.2105334594&width=266&height=312&newmaps=0&reproj.automerc=1&num=5&delay=25&timelabel=1&timelabel.y=10&rainsnow=1&smooth=1&radar_bitmap=1&xnoclutter=1&xnoclutter_mask=1&cors=1
google map base url: http://maps.googleapis.com/maps/api/staticmap?center=49.2381322,17.6717151&zoom=11&size=266x312&maptype=hybrid&markers=color:red|size:small|49.2381322,17.6717151
radar url: http://api.wunderground.com/api/'remove number'/animatedradar/lang:CZ/image.gif?maxlat=49.3080209651&maxlon=17.7630389525&minlat=49.1681443993&minlon=17.5803912475&width=266&height=312&newmaps=0&reproj.automerc=1&num=5&delay=25&timelabel=1&timelabel.y=10&rainsnow=1&smooth=1&radar_bitmap=1&xnoclutter=1&xnoclutter_mask=1&cors=1
google map base url: http://maps.googleapis.com/maps/api/staticmap?center=49.2381322,17.6717151&zoom=6&size=311x398&maptype=hybrid&markers=color:red|size:small|49.2381322,17.6717151
radar url: http://api.wunderground.com/api/'remove number'/animatedradar/lang:CZ/image.gif?maxlat=52.0110045448&maxlon=21.0884631469&minlat=46.3003600414&minlon=14.2549670531&width=622&height=796&newmaps=0&reproj.automerc=1&num=5&delay=25&timelabel=1&timelabel.y=10&rainsnow=1&smooth=1&radar_bitmap=1&xnoclutter=1&xnoclutter_mask=1&cors=1
google map base url: http://maps.googleapis.com/maps/api/staticmap?center=49.2381322,17.6717151&zoom=10&size=311x398&maptype=hybrid&markers=color:red|size:small|49.2381322,17.6717151
radar url: http://api.wunderground.com/api/'remove number'/animatedradar/lang:CZ/image.gif?maxlat=49.4162422575&maxlon=17.8852618529&minlat=49.059377517&minlon=17.4581683471&width=622&height=796&newmaps=0&reproj.automerc=1&num=5&delay=25&timelabel=1&timelabel.y=10&rainsnow=1&smooth=1&radar_bitmap=1&xnoclutter=1&xnoclutter_mask=1&cors=1
getting current and forecast:Sat Oct 29 11:00:10 2016
http://api.wunderground.com/api/'remove number'/conditions/astronomy/hourly10day/forecast10day/lang:CZ/q/49.2381322,17.6717151.json?r=0.531317309923
wxstart for radar1
get radar api call spacing oneshot get i=0.382641077042
wxstart for radar2
get radar api call spacing oneshot get i=2.38017320633
getting radar map radar1:Sat Oct 29 11:00:11 2016
getting radar map radar2:Sat Oct 29 11:00:13 2016
radar map received:radar1:Sat Oct 29 11:00:20 2016
radar map frame count:radar1:5:r0
radar map received:radar2:Sat Oct 29 11:00:21 2016
radar map frame count:radar2:5:r0
Terminated`

When I try insert one of this url to web browser, output is here:
`
{
  "response": {
  "version":"0.1",
  "termsofService":"http://www.wunderground.com/weather/api/d/terms.html",
  "features": {
  
    "animatedradar": 0
    ,
    "radar": 0
  }
				,
	"error": {
		"type": "radarargs"
		
			,"description": "you must supply query arguments for the radar image."
			,"examples": ["maxlat=48.3&maxlon=-68.8&minlat=35.3&minlon=-92.6&width=1085&height=798&rainsnow=1&timelabel=1&timelabel.x=925&timelabel.y=41"]
		
	}		
	}`

n0bel, 2016-10-29:
The radar is "trying" to work fine.

This line shows it is: 
radar map frame count:radar1:5:r0

Its saying that radar1 has 5 frames with 0 retries.

You'll also notice that the time is incrementing as it should.

When I tried your URL from the log it worked fine, no errors.  You must not have copied the entire URL.

The real problem is that Weather Underground doesn't always have radar data for all parts of the world.  Several people have reported this to me -- (look at some of the past issues).    This is why I added the satellite option.  It seems they always have cloud cover/cloud height information.   Give that a try.

GeorgeJirka, 2016-10-29:
Ok. Thank you for your answer.

closed 2016-11-02 by n0bel