n0bel/PiClock #44: Issue when starting PiClock

UserResu1234, 2016-11-29:
Disabling screen blanking....
Setting sound to max (assuming Monitor Tv controls volume)....
Checking for NeoPixels Ambilight...
Starting NeoPixel Ambilight Service...
Checking for GPIO Buttons...
Checking for Temperature Sensors...
Starting PiClock.... logging to screen.
QGtkStyle was unable to detect the current GTK+ theme.
Traceback (most recent call last):
  File "PyQtPiClock.py", line 173, in wxfinished
    wxdata = json.loads(wxstr)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
getting radar map radar1:Tue Nov 29 15:38:35 2016
get radar error radar1:1
getting radar map radar2:Tue Nov 29 15:38:37 2016
get radar error radar2:1
^CTraceback (most recent call last):
  File "NeoAmbi.py", line 45, in <module>
    sleep(0.2)
KeyboardInterrupt

I'am always getting this Problem when I'am starting the PiClock. Does somebody can help me because I'am really stuck at this Problem.

dankolbrs, 2016-11-29:
@UserResu1234, Have you double checked you've got a valid API key in `./Clock/ApiKeys.py`? This may be from an invalid API key being used and the API returning something like a 400 rather than the expected JSON.

You should be able to log in to your WUnderground account and verify you're not over limits, and get a new API key if you want to swap it to verify.

UserResu1234, 2016-11-30:
The API key is valid. I checked this, even tried a new one. This did not work for me.

dankolbrs, 2016-11-30:
I wonder if this is like #43, 

There should be URL's given in the Clock/PyQtPiClock.{#}.log. Using the URLs given there, you can copy/pasta that into your URL bar and see what is actually returning from those calls.

Something like below would be expected (though with your lat/lon and key from ApiKeys.py and Config.py). The number after `r=` will be be different for each logging of it.
```
http://api.wunderground.com/api/{KEY}/conditions/astronomy/hourly10day/forecast10day/lang:EN/q/{LAT},{LON}.json?r=0.127276964802
```    
and:
```
radar url: http://api.wunderground.com/api/{KEY}/animatedradar/lang:EN/image.gif?maxlat={LAT}&maxlon={LON}&minlat={LAT}&minlon={LON}&width=664&height=597&newmaps=0&reproj.automerc=1&num=5&delay=25&timelabel=1&timelabel.y=10&rainsnow=1&smooth=1&radar_bitmap=1&xnoclutter=1&xnoclutter_mask=1&cors=1
```

n0bel, 2016-11-30:
It is strange that the logs provided in your original post do not include the urls for weatherundground.   Perhaps you removed them to keep the api keys private, but that is exactly where you should go first. 

If the urls don't print, perhaps you're using a very old version of the PiClock?

UserResu1234, 2016-12-01:
The URLs are printing, I just removed them to keep my keys private. I checked the URL which gives me the forecast. My Browser shows a large Text with several Forecasts etc. The radar URLs also returning animated Forecasts in my Browser. So I think this isn't the Problem. I don't get it. There is a correct JSON File, all the animated radars and it's still returning this ValueError.

So when I lock at the PiClock when it's started I get the correct time and date, but the two boxes for the maps are just grey and the all boxes for the weather are empty. PiClock only shows me the Clock and the Date.

git pull says that my Version of PiClock is the newest one.

n0bel, 2016-12-20:
Did you ever find a solution for this? I don't have an answer, and I've not seen this before.   If the JSON shows in your browser, then it should be working on PiClock.   Perhaps you could send my the URL (without your key) in a email to me.

n0bel, 2017-01-07:
Any further update on this one?   I'm unable to reproduce the issue.

zeroadam, 2017-07-25:
Traceback (most recent call last):
  File "PyQtPiClock.py", line 9, in <module>
    import json
  File "/usr/lib/python2.7/json/__init__.py", line 1
SyntaxError: Non-ASCII character '\xe3' in file /usr/lib/python2.7/json/__init__.py on line 1, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

I recently started getting this in the log. PiClock has been running great for months until this. I have checked WU apikey and ensured git branch is current.

n0bel, 2017-07-25:
I'm guessing, but I'd say that your SD card is corrupt.   The error is pointing to a pre-installed library.

zeroadam, 2017-07-26:
Yes, I thought that was weird that I was getting an error in python. I thought about trying uninstalling and reinstalling python, but easy enough to just redo the rpi card though!

closed 2018-12-23 by n0bel