n0bel/PiClock #45: Different issue with starting clock

3l3phant, 2016-12-17:
I'm getting a different error message than 44. I'm wondering if the order in which I installed things might be causing the problem. Running Sierra 10.12.1 (16B2555):

Traceback (most recent call last):
  File "PyQtPiClock.py", line 14, in <module>
    from PyQt4 import QtGui, QtCore, QtNetwork
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/PyQt4/_qt.so, 2): Library not loaded: QtDesigner.framework/Versions/4/QtDesigner
  Referenced from: /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/PyQt4/_qt.so
  Reason: unsafe use of relative rpath QtDesigner.framework/Versions/4/QtDesigner in /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/PyQt4/_qt.so with restricted binary

dankolbrs, 2016-12-19:
I can't confirm as I don't have a Mac, but a search that brought up a [stackoverflow question/answer looks like this is due to Apple's System Integrity Protection.](http://stackoverflow.com/questions/31343299/mysql-improperly-configured-reason-unsafe-use-of-relative-path#31821332) Looks like this is an issue in the PyQt libraries and Mac, not one of PiClock specifically.

There is a solution that describes how to change a relative import to an absolute, if you want to go that route. I don't think there'd be an issue to it as its specifying an absolute path to a binary, and that doesn't seem to be a large or dangerous change to your system. If you're familiar with python virtual environments (I think you'd want to have something like brew installed to use which could make large system changes), that may be another way to go.

n0bel, 2016-12-20:
I'm glad somebody recognized python on a mac, because I knew it surely wasn't a pi or windows with those path names.   I'm hoping the community or @dankolbrs can help resolve this, because I don't have access to a mac.  Thanks!

dankolbrs, 2016-12-20:
No Mac here to help troubleshoot or test the solution given in the stackoverflow link in the previous comment, unfortunately.

3l3phant, 2016-12-21:
Odd. The file, "libQtCLucene.4.dylib" isn't anywhere on my system. I'm all
the way down in the
"/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/PyQt4"
directory, but the file isn't there.

3l3phant, 2017-01-07:
> Has this been figured out? I don't have access to a mac to help
> troubleshoot.

TBH, life got in the way of my being able to come back to test :(

closed 2017-04-18 by n0bel