n0bel/PiClock #46: PiClock Touchscreen

nerdhrdr, 2017-01-04:
Hello,

I used a touchscreen for my display.  I am trying to make it so a tap changes from the Clock screen to the Radar and back.

I found in PyQtPiClock.py where you have it set for the spacebar and arrown keys.  I am hoping you can help me with the if statement for the tap.

Thanks,

dankolbrs, 2017-01-06:
From what I read in the docs, looks like you'd need to add a new callback function for mousePressEvent() as keyPressEvent() only grabs keyboard interrupts.
http://pyqt.sourceforge.net/Docs/PyQt4/qwidget.html#mousePressEvent

It can be siimilar to the [keyPressEvent function](https://github.com/n0bel/PiClock/blob/master/Clock/PyQtPiClock.py#L677-L697) already in use. Depending on the touchscreen I would assume it would just provide left click event, and it would just be a matter of checking for that and having some kind of toggle flag to determine whether to go forward or back.

nerdhrdr, 2017-01-07:
That worked great.  If any one is interested I added the following lines to the MyMain class in PyQtPiClock.py to get the screen to change on type or mouse click:

 def mousePressEvent(self, event):
        if type(event) == QtGui.QMouseEvent:
				nextframe(1)

n0bel, 2017-01-07:
I'm reopening the issue to add it as an enhancement so I don't forget to add this to the base code.

GREAT WORK you guys.. I love it when people add things to this project.

dankolbrs, 2017-01-07:
@n0bel Added a [PR](https://github.com/n0bel/PiClock/pull/48)

@nerdhrdr, You got dibs on this one. I plugged in my touchscreen to fire this up today and saw you already had the fix. Feel free to submit a PR (if you're interested in it) and I'll close mine.

nerdhrdr, 2017-01-07:
Fell free to go with yours.

hsoj95, 2017-01-08:
Wow, that works perfectly nerdhrdr! I added it to mine and I can say it works great on the default Raspberry Pi touchscreen! Thanks for figuring that out! 😃

n0bel, 2017-01-22:
check check... updated in commit https://github.com/n0bel/PiClock/commit/1b39b08b5b11371bd6143487e3b49506a9580d7d

closed 2017-01-22 by n0bel