n0bel/PiClock #47: Ability to have the background image change based on weather conditions

hsoj95, 2017-01-06:
I love PiClock and have been using it on my mounted touchscreen Raspberry Pi since the beginning!

A feature I thought would add a lot to it is the ability to have the background of the PiClock change images based on the current weather conditions. It might add a really nice dynamic effect to the program. It might even be possible to use a video as the background, but that might cause processing issues on weaker Pi's.

I included a few links to some copyright free images from Pixabay for a few of the categories used in the program! Hopefully you'll find them to be of use! 😃 

Sunny:
https://pixabay.com/en/sunrise-mountain-nature-light-1922408/

Partly Cloudy: 
https://pixabay.com/en/sky-clouds-sun-rays-weather-blue-1365325/
https://pixabay.com/en/cloud-sky-sky-clouds-blue-white-1601928/

Cloudy:
https://pixabay.com/en/storm-clouds-clouds-thunderstorm-426271/

Rain:
https://pixabay.com/en/rain-after-the-rain-a-drop-of-455120/

Thunderstorm:
https://pixabay.com/en/lightning-storm-night-firebird-342341/
https://pixabay.com/en/lightning-storm-weather-sky-399853/

Snow:
https://pixabay.com/en/runners-head-winter-69657/

n0bel, 2017-01-07:
Sounds like a great possibility.  Some of them will need to be cleaned up so as not to wash out the wording on the page.   If anybody wants to do this before I get to it, I do consider all pull requests!

hsoj95, 2017-01-08:
Hmm, I will see what I can do! I'm not a Python programmer by any stretch of the imagination, so actually doing the code is sorta out of my league, However, I will try and get the images cleaned up a bit, or maybe find some better suited ones instead. If the text becomes an issue, one possibility might be to have the text-color change depending on what image is used. A darker color text for brighter images, and a brighter text for darker images.

hsoj95, 2017-01-20:
Having looked at the code a bit after cloning it, I do have one major question. Where do you set and control the background image? Is it in the main python file? Because I was not able to ascertain where the background image is set. I assume it's where the main Qt frames are set, but I couldn't specifically see where that was controlled. 

Any suggestions or hints as to how this works would help me a lot! Being that I'm not a Python guy, I think I can create this feature, just need a bit of a helping hand to start out with! 😃

karlLayton, 2017-01-20:
It is in the Config file

n0bel, 2017-01-22:
Line 811...  and below.... where it creates a string with
```
......border-image: url(" + Config.background + ")........
```
That is what sets the frame's background image.   There are two frames one for the first page and one for the second, as you can see they are set the same way...    If you want to change them later, you might have to reference the frames array   frames[0] and frames[1] with the setStyleSheet instead of frame1 and 2 depending on the context of your call.

```
frames = []
framep = 0

frame1 = QtGui.QFrame(w)
frame1.setObjectName("frame1")
frame1.setGeometry(0, 0, width, height)
frame1.setStyleSheet("#frame1 { background-color: black; border-image: url(" +
                     Config.background + ") 0 0 0 0 stretch stretch;}")
frames.append(frame1)

frame2 = QtGui.QFrame(w)
frame2.setObjectName("frame2")
frame2.setGeometry(0, 0, width, height)
frame2.setStyleSheet("#frame2 { background-color: blue; border-image: url(" +
                     Config.background + ") 0 0 0 0 stretch stretch;}")
frame2.setVisible(False)
frames.append(frame2)
```

merky1, 2017-08-18:
I wrote an external perl script to switch configs based on temps.

`use WWW::Wunderground::API;
use Switch '__';

# location
my $wun = new WWW::Wunderground::API('20147');

my $wun = new WWW::Wunderground::API(
  location => 'Ashburn, VA',
  api_key => '70c328e8a8c94763'
);

$t1 = $wun->api_call('conditions');
#print "Temp_f:".$wun->temp_f."\n";

my $temp = $wun->temp_f;

switch ($temp) {

    case __ < 30 { `sh /home/pi/PiClock/switcher.sh Config-cold` }
    case __ < 40 { `sh /home/pi/PiClock/switcher.sh Config-30s` }
    case __ < 50 { `sh /home/pi/PiClock/switcher.sh Config-40s` }
    case __ < 60 { `sh /home/pi/PiClock/switcher.sh Config-50s` }
    case __ < 70 { `sh /home/pi/PiClock/switcher.sh Config-60s` }
    case __ < 80 { `sh /home/pi/PiClock/switcher.sh Config-70s` }
    case __ < 90 { `sh /home/pi/PiClock/switcher.sh Config-80s` }
    else         { `sh /home/pi/PiClock/switcher.sh Config-hot` }
}
`

n0bel, 2017-09-04:
great idea @merky1

closed 2017-09-04 by n0bel