n0bel/PiClock #49: Progress bar for opening PiClock

hsoj95, 2017-01-10:
I was trying to create a progress bar that showed how long the user had until the program launched when the desktop icon was clicked. I was using the zenity --progress interface to try and achieve this. However, I am having difficulty getting the window to function properly. 

I used the simple tutorial [here](https://help.gnome.org/users/zenity/stable/progress.html.en) for using the zenity progress window. Here's how the code looks currently:

`if [ "$1" = "-m" -o "$1" = "--message-delay" ]
then
        MSG="echo Waiting $2 seconds for response before starting"
        #DELAY="xmessage -buttons Now:0,Cancel:1 -default Now -timeout $2 Starting PiClock in $2 seconds"
        #DELAY='zenity --question --title PiClock --ok-label=Now --cancel-label=Cancel --timeout '$2' --text "Starting PiClock in '$2' seconds" >/dev/null 2>&1'
        DELAY='(
        echo "10" ; sleep 1
        echo "20" ; sleep 1
        echo "30" ; sleep 1
        echo "40" ; sleep 1
        echo "50" ; sleep 1
        echo "60" ; sleep 1
        echo "70" ; sleep 1
        echo "80" ; sleep 1
        echo "90" ; sleep 1
        echo "100" ; sleep 1
        ) |
        zenity --progress --title="Starting PiClock" --text"Preparing to start PiClock in 15 seconds" --percentage=0'
        shift
        shift
fi`

That seemed the closest to achieving what I was aiming for, but it still wasn't functioning right, namely it just launched the PiClock instantly. I am not very familiar with how zenity actually functions, so if either of the two main contributors knows how to make this function, I would love to add the functionality to the program!

dankolbrs, 2017-01-10:
@hsoj95 

For the formatting question, use ` ``` ` at the beginning and end of your text block for the script. Looks like you may have just used a single ` at the beginning and end. I tend to liberally use of the Preview tab when entering/editing comments as well. :D

[Github Markdown Help](https://help.github.com/articles/creating-and-highlighting-code-blocks/)

hsoj95, 2017-01-11:
Awesome! That worked great! Thanks for the tip about the triple things! :D

n0bel, 2017-01-22:
Great to see peeps working on this together... THANKS!

closed 2017-01-22 by n0bel