n0bel/PiClock #56: Error Handleing

pjmcharr, 2017-02-10:
Every now and then PiClock will crash. Is there any error handling that is available or would a cron job to start PiClock in this case be the best fix?

n0bel, 2017-02-12:
There is some error checking, especially with the gif radar images that wu seems to send badly formatted sometimes.     To see why it is crashing, look at the log files in PiClock/Clock folder.

rcomito, 2017-02-15:
Hi pjmcharr - I have a script that checks the clock status every 5 minutes under control of cron.

I can't seem to get the "insert code" to work, so it's not very pretty, but you get the idea.

PIDs=$(ps -ax | grep -c "PyQtPiClock.py")
if [ $PIDs -ne 2 ]l then        # One of these is the "grep"
    if [ $PIDs -gt 1 ]; then
        sudo killall PyQtPiClock
    fi
    sh /home/pi/PiClocl/startup.sh -n
fi

Here's the cron entry:
*/5 * * * * /home/pi/bin/chkclock

Hope that helps.

Rick Comito

pjmcharr, 2017-02-16:
Awesome thanks for the help guys!

n0bel, 2017-02-16:
I've had no need to check if it crashes.   However I do a reboot via crontab every day at 3am.   That said, I run several that simply never crash.

closed 2017-02-25 by n0bel