n0bel/PiClock #58: NOAA audio feeds issues

CigarManDan, 2017-03-05:
Is the function for the audio feeds not working for everyone or are some of us doing something wrong?

n0bel, 2017-03-18:
It works for me.  Check the URL.  Check if it works on your pc.   Check if the player command works..
mpg123 -q http://yourstreamurl  from the command line.

n0bel, 2017-04-18:
At this point the weather underground noaa feeds seem to be mostly down.   I'm not sure what to do for a replacement.

closed 2017-04-18 by n0bel

n0bel, 2017-06-24:
NOAA weather radio streams have reappeared on the net. You can find it here: http://noaaweatherradio.org/ They don't put the .mp3 urls where they are easily accessable, so you need to use your browser to "View Page Source" in order to find the proper .mp3 url.

hsoj95, 2017-06-25:
There's still many areas lacking coverage. I live in Georgia, and there's literally no coverage for my state at all. Same for large parts of Florida. I guess there's just some stations that just post their streams online.

n0bel, 2017-06-25:
As with the weather underground streams, these are private companies or people donating receiving equipment, power and bandwidth to provide this service.  All I can do is provide this information as I come across it.   None of these streams are provided by the government.