n0bel/PiClock #59: weather underground

maserowik, 2017-03-09:
Very nice project.  
I am having a slight configuration issue.

i have my APO for both Google and Weather Underground set up itne the ApiKeys.py
i am getting valid weather output but on the lower left wherne teh wu images are i see the clouds but not the outline of my location
Please look at the attached screen capture.
Thanks
Mike

[image]

maserowik, 2017-03-09:
found the problem.
it was that i did not enable the api in the dashboard.
once done it works.

closed 2017-03-09 by maserowik

cswanger314, 2018-01-11:
How do you enable the API in the dashboard? I am having the same issue. 

Thanks,

n0bel, 2018-01-21:
Just leave the google API key empty.   As the instructions say, it is not required to make it work, because the volume is so low.