n0bel/PiClock #60: Weather Warnings

maserowik, 2017-03-09:
Is there a way to add Weather Warning/Alerts/Watch to the WX screen?

dankolbrs, 2017-03-09:
Unfortunately, not without significantly modifying the code. I assume the Weather API has some return for active alerts, but I'd have to take a look. This would require pulling that from the API, and adding a QT widget in the display to display the alert/warning/watch.

I think it would be an awesome feature, and I'd bet that @n0bel tags this with an `enhancement` label. Also, I'm tagging issue #54 to link this up with that discussion.

n0bel, 2017-03-11:
Sounds like something worth while. It'll make it into the refactor somewhere.

https://www.wunderground.com/weather/api/d/docs?d=data/alerts

closed 2018-12-23 by n0bel