n0bel/PiClock #62: Add second data feed to 7 Day Forecast

VTHokiEE, 2017-03-15:
I'd like to add a second data feed to the 7 day forecast (for my purposes I'd like to switch between the weather and a surf report every couple seconds). I'm having issues separating wxfinished() from getwx(), it seems like the json string disappears if I move wxreply.finished.connect(wxfinished) to a different function. Can you offer any insight as to what is going wrong? This seems like it would be a nice enhancement to allow further customization of those windows.

n0bel, 2017-03-18:
no idea without seeing your mods.

VTHokiEE, 2017-03-20:
I figured it out, but it certainly looks like a hack to my eyes. I'll try to upload an example shortly.

closed 2017-04-18 by n0bel