n0bel/PiClock #69: Clone the git

cwfuselier, 2017-03-28:
When I follow he steps to clone the PiCLock it ask for a username & password.

n0bel, 2017-03-28:
If you miss type any part of the URL that is the response you get.

```
git clone https://github.com/n0bel/PiClock.git
```

I tried for example an incorrect url:
```
[kevinu@cricket2 ~/jjjtest]$ git clone https://github.com/n0bel/PiClick.git
Cloning into 'PiClick'...
Username for 'https://github.com': 
```

So there you go.. check the spelling.. better yet do a copy/paste if you can.

cwfuselier, 2017-03-28:
ok thanks will try it. Now on the reinstall from fresh the install python-qt4 givea a error when starts to try to download. says 0%[Waitning for headers] then says Connection failed for everything it trys to install. It did the install update fine just this one now.

n0bel, 2017-03-29:
That message occurs when the Raspberry Pi Repositories are down.   If you try later, it'll probably work.

n0bel, 2017-03-29:
When this happened to me, i found this post helpful.  https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=126355

brandondasdf, 2017-04-11:
I really like this project it is very cool. I've been trying to install it a couple times now. 

When I run the 
git clone https://github.com/n0bel/PiClock.git

It loads an then stops at 45%. Any idea why this is happening?

n0bel, 2017-04-12:
If it does this repeatedly, it is either power or the sd card.    Start with the SD card be sure it is class 10 or uhs.

brandondasdf, 2017-04-12:
Okay thanks mate I will try that. Could you perhaps elaborate on the class
10? Is that the formatting?

n0bel, 2017-04-12:
https://www.howtogeek.com/189897/how-to-buy-an-sd-card-speed-classes-sizes-and-capacities-explained/

brandondasdf, 2017-04-13:
Thank you again

closed 2017-04-18 by n0bel