n0bel/PiClock #75: PiClock2: Have you considered building version 2 in Java?

hsoj95, 2017-05-25:
Recently, I was goofing around with Java on my Pi and I made a fairly simple and easy to read digital clock for my Pi using Java Swing. A key point I learned is to multithread the code so it doesn't use all of the CPU power, since the Pi v3 has a quadcore processor. Doing this meant that the digital clock runs extremely well on just 25% processor power. This led me to an idea though, would PiClock2 run better if it was built in Java using multithreading? I don't know if @n0bel knows Java or not, but it might a more efficient alternative to Python with Qt. I would certainly be willing to help as, next to HTML, CSS, and JavaScript (Web Programming), Java is one of the languages I know best!

(I would do the superscripting for the "2" in the title, but I don't know how! The MarkDown instructions didn't say how to do that. 😕 )

hsoj95, 2017-05-25:
To follow up on what I suggested I thought I'd add a few things. First, the Java program could either use Swing (easier to work with, but maybe not as customizable as you'd like) or JavaFX (More complicated, but the stars are the limit in terms of customizations). Secondly, from what I have heard, Java seems to run fairly well even on the older Pi's. This might solve the issue of the Clock not running smoothly on older hardware. Although I'm sure it would ultimately use up more processing power on older hardware, ultimately this will be true no matter what language you build it in!

dankolbrs, 2017-05-26:
I know you're not asking me this, but you also know what they say about opinions. :wink: 

In any case, python has multi-threading, and there is even the [QThread](http://doc.qt.io/qt-5/qthread.html) library in PyQt which could handle this. The problem would be that older hardware (thinking RPi1s, Zeros, the P3 on my shelf, the Celeron laptop that I actually run PiClock on...) wouldn't benefit from multi-threading, and the extra overhead of multi-threading on a single thread CPU would actually make it worse. Threading is only beneficial to about the number of threads available on the CPU, then it hurts performance due to the extra overhead. My understanding is that Java is better with multi-threading than Python, but I would imagine it would be negligible in this application even with an RPi3. For UI performance PyQt already has a benefit of being event driven so things that take some time (such as downloading radar images) don't block the UI, but I can't speak to whether or not Swing/JavaFX are event driven as well.

I think the amount of people that use python, the ease of reading the code, the amount of documentation, and the ease of package installation via pip is probably enough to justify python over Java. I'm not going to slam on Java, it has its merits and some amazing projects are written in it. And hey, it could be a good personal project for a weekend, and you might get someone like me lurking on your Github and providing their opinions when someone asks you a question. :smile: 

Since you're using Swing, you may be interested in [qt-designer](http://doc.qt.io/qt-5/qtdesigner-manual.html). It's a drag and drop IDE for QT. I've installed it from the standard repos on Fedora and Ubuntu. I've started messing around with it, using it to add elements which saves as a `*.ui` file, you can convert that to python with `puic5 form.ui -o myform.py`. There's plenty of things I haven't touched with it to make it even easier, like probably handle events and such, but being able to see what you're creating is already a huge plus. I was even playing around making a UI for the PiClock config to see how that would work. I haven't put it up here yet since I got distracted the last time I planned on doing something specific here, and I got distracted after messing with it anyway, but I'd be happy to put it up you want to take it and run with it to get to know python/pyqt better.

n0bel, 2017-05-27:
PiClock could be written in virtually any language connected to virtually any GUI. Originally it lived in a browser. http://n0bel.net/v1/index.php/projects/raspberry-pi-clock/8-raspberry-pi-clock-how-to-waste-a-saturday-afternoon  Then it became an exercise for me to do something useful with python, (at the time the language of choice for Pi users, for reasons I still have no clue about)  http://n0bel.net/v1/index.php/projects/raspberry-pi-clock/30-goodbye-piclock-hello-pyqtpyclock     I won't be moving away from Python for this project.   If I changed anything to do with the basic architecture of PiClock it would be to move away from Qt and to Kivy, but of course still with a python base.   My tests with Kivy show a huge performance difference, since it uses graphics acceleration (OpenGL).  The primary performance issue for PiClock is graphics.   Writing moving hands over the background every second and writing the radar images with map underlay and marker pins overlay several times a second are the primary performance issues.   Moving this to the GPU (read as shaders) (which Qt5 doesn't do) would give the performance boost.  Even with all this said, a PiClock running on an old original Pi, consumes about 60% of the CPU.   Pi2 and Pi3 it sits at 20 to 30%.  Is there a problem?  I don't see one.   Threading won't fix the single threadedness of the display buffer.  I don't really understand the statement that threads lowered the cpu bandwidth in the described clock app.   Doing the same work spread over threads is less efficient than in one thread, but the total work gets done faster (parallelism if multiple cores are available).  The total cpu cycles is still the same or worse.  I'm probably missing some aspect of what you were doing with the app.

hsoj95, 2017-05-27:
@dankolbrs Your opinion is always welcomed! :smile: 
I actually didn't know python had multi-threading, although knowing it complies to C, I should maybe have guessed that. I also did forget that the older Pi's only had single core CPU's on board, which means my idea for using multi-threading wouldn't work. To answer your question though, both Swing and JavaFX are event driven! My main reasoning for thinking Java might be better was the thought that it might provide better performance than Python would. However, knowing the Pi community is built strongly around Python more than Java for sure, a fact I will rant about down below in a bit, it probably is best that it stay in Python. That's interesting QT has a drag and drop IDE for it! Java actually has something similar through NetBeans! Makes it SOOO much easier to work with Swing than having to code 400 lines of code by hand!

@n0bel That's interesting it was originally a browser program! I remember seeing that on your blog back when I first discovered PiClock late last year. Any chance you'd ever upload those browser files to your GitHub? I'd love to dig into them and have a look! :wink: 
I can understand why you moved it to Python. Python is still basically the goto language for coding on the Pi, a practice I am overall not happy with. Python is an ok language, but I don't believe it's the best language to use, certainly not for beginners learning programming. I can understand why you don't wanna move the project to a different language though, at least for the second incarnation of it. The amount of work alone porting it would be daunting! When you suggested Kivy, I went out and experimented with it a bit, and I must say, I actually quite like it! It has a fairly clean and easy to understand code base, as opposed to QT that is definitely not the easiest thing to learn. 
My idea for using multi-threading in Java was the idea that you could have one thread working the clock part of the clock, and maybe the changing backgrounds based on the weather \*hint\*:wink:, and another thread for gathering the weather data and running the radar gif. On the Pi3 this would make the CPU top out at 50%, but in theory that 50% would result in a butter smooth performance of the PiClock. It might even allow for some overhead room to add extra features. But like what @dankolbrs said up above, I forgot the earlier Pi's only had single core CPU's, which would make my idea basically fall flat.

I will say though, don't totally give up on the idea of using another language for it in the future, especially for PiClock3! (Yes, I'm that forward thinking!) A language like Java, Ruby, or even HTML/CSS/JS might prove to be great alternatives in the future, especially if more powerful Pi's are released! Other languages may prove to be useful in the future as well. I know Apple's Swift is making great strides in becoming a web based language. Apple is also making strides in Swift being compatible with Linux. So that might be a language that proves to be useful in the future as well! In any case, let's get PiClock2 going strong! I am not great with Python, but I'll help however I can! :smiley:

hsoj95, 2017-05-28:
@n0bel Actually, thinking back to it, I'd be highly interested in looking at the original browser program! I might be able to improve the code and add in the features currently available in the Python version. This thought occurred to me when I was trying to get the PiClock to run on my Mac, and I discovered it was less than a simple task to install PyQT on my Mac. If I could get the browser program running efficiently, it might make a great, and easier to use, alternative to the Python version for people who wanna run this on something other than a Pi! It at the very least would be a lot easier to download and use for non-Pi users!

hsoj95, 2017-06-10:
Whoops! I meant to close this thread after my last reply!

closed 2017-06-10 by hsoj95