n0bel/PiClock #78: Automatic Switching Between Primary and Secondary Screen

ChrisBarnesInfo, 2017-06-04:
First of all - THANKS! This is a cool app and I can tell a lot of people have put time into this. So... thanks!

Now - I'm setup and working just fine. I chose to have no external inputs (no LEDs, no buttons, etc). I plan on running this with just a Pi and a monitor - no keyboard, no mouse.

How can I get a timed alternation between the primary screen and the secondary screen (with the larger radars). Since people viewing this display won't have the ability to manipulate it I'd like some kinda of automatic "press space" or "click mouse" function to alternate between these 2 screens every X seconds.

Thanks!

hsoj95, 2017-06-04:
In theory, this shouldn't be too hard to accomplish with a loop calling the command to switch screens. One thing I would worry about is that this could cause too many calls to be made to the WU api server. Since each screen makes it's own update call, I'm not sure whether doing this might cause too many calls to be made. @n0bel will know this for sure! It's a great idea nonetheless @ChrisBarnesInfo! 😄

n0bel, 2017-06-04:
> One thing I would worry about is that this could cause too many calls to be made to the WU api server. Since each screen makes it's own update call, I'm not sure whether doing this might cause too many calls to be made. @n0bel will know this for sure!

Yep it would.

However in PiClock² (the dev branch)  @dankolbrs has added caching to avoid this issue.

Also @gshafer just posted his solution for no buttons/keyboard/mouse (in issue #37) https://youtu.be/zyORXt4QIc0

hsoj95, 2017-06-04:
>However in PiClock² (the dev branch) @dankolbrs has added caching to avoid this issue.

Ah, excellent! Glad to know y'all were thinking ahead for this issue!

ChrisBarnesInfo, 2017-06-05:
Thank you all for the quick response. I understand the API call issue and will hold off on the alternating screens on a timer approach.

closed 2017-06-05 by ChrisBarnesInfo

gshafer, 2017-06-14:
Added gesture control using the Flick hat. Fun addition. Here is the video:

https://youtu.be/HgrOASov0N0