n0bel/PiClock #80: No weekly forecast

bigbadmayo, 2017-08-29:
I have a satellite image but no radar the weekly forecast is blank. I've tried flashing and reinstalling as well as checking WU's API and lat lon in Config.txt. This Raspberry Pi 2 b is also running Pi-Hole. I have had this set up running before changing from Jessie to Stretch. This is my first issue and I can't wait to get it up and running again. Let me know what files you need!

n0bel, 2017-09-04:
Check the log files (in the Clock folder) for errors, or other information.  Also included is the url for the forecast and the radar, so you can copy/paste to try in your browser to diagnose.   PyQtPiClock.log is the latest one.

It is most likely an API key, or a lat lon issue.

bigbadmayo, 2017-09-04:
What is the log file called in Clocks?

n0bel, 2017-09-05:
PyQtPiClock.log is the latest one.

bigbadmayo, 2017-09-06:
I have a PyQtPiClock.py but no .log

bigbadmayo, 2017-09-14:
My brilliant self was using the wrong WU Api.
Thanks for the help!

closed 2017-09-14 by bigbadmayo