n0bel/PiClock #81: Opacity rain on radar images

TQ255, 2017-09-14:
Hi,

is it possible to change the opacity of the rain on the radar display? It now has no opacity at all so you can't see the roadmap.

Here is an example:

https://imgur.com/4bS6eMA
All rain and no idea where it is located.

Here is an example what i want to achieve:
https://imgur.com/a/HC0Yf
The opacity has changed and the roadmap becomes visible again.

SerBrynden, 2017-09-15:
I modified my PiClock to use Mapbox with OpenStreetMap instead of Google Maps.  With Mapbox, I created a base layer for land/water and a layer for all the roads, labels, and borders.  Then the weather radar goes in between those two map layers.  I'll share my modified python files soon, once I clean them up a bit.

TQ255, 2017-09-18:
Cool, thanks SerBrynden.

SerBrynden, 2017-09-25:
@TQ255, I created a fork of PiClock, which should have what you're looking for.  It's not yet ready to be merged with n0bel's original PiClock, but it's the version that I use.  To use it, just pull [my version](https://github.com/SerBrynden/PiClock) and do a fresh install.  As the instructions say, you'll need to get a Mapbox API key, which is free, and put it in the ApiKeys file.

n0bel, 2017-09-27:
If I recall Qt, It should be a few lines of code to change the opacity of the radar overlay.

@SerBrynden Thanks for sharing your mods!

TQ255, 2017-10-06:
That's much better, i also like that the maps are darker. Thanks!

closed 2018-12-23 by n0bel