n0bel/PiClock #83: HamClock

dankolbrs, 2017-09-22:
Sorry, not really a PiClock issue but I thought this was cool and wanted to share. 

I don't know if [this](http://clearskyinstitute.com/ham/HamClock/) caught anyone else's eye in the latest issue of QST. The code is available but I didn't see a mention of license. The project is Arduino based, and requires some specific Arduino libraries but could probably be ported to a Pi by someone interested enough. The weather aspect is a BME280 sensor. 

Built by WB0OEW, more info, code, and full article at http://clearskyinstitute.com/ham/HamClock/
[image]

SerBrynden, 2017-09-22:
As an RF engineer working on a NOAA contract, I love it!  Thanks for sharing!

hsoj95, 2017-09-23:
This is so cool!

n0bel, 2017-09-27:
@dankolbrs thanks for sharing.   Could the next incarnation of PiClock have the flexibility and plugin-ability, in order to come up with this type of screen, or pieces of it?  I'm hoping so.

skipwill, 2023-04-11:
,I have both hamclock and PiClock running. PiClock is on a 32in tv monitor on a pi A and hamclock is on a pi w VNCed to an old 10in iPad.