n0bel/PiClock #85: Background change

RyUp1, 2017-09-25:
Hello! I am super new to all of this but I just finished your build! It was great, thanks. How can I change the background image? Also, I saw you added a digital style clock but I could not find a guide on how to change to that. Sorry, I am still learning but this has helped a ton! Thanks!

dankolbrs, 2017-09-25:
Hi @RyUp1 

The background image can be changed by updating the Config.py `background` field to point to the image you want to use. 
https://github.com/n0bel/PiClock/blob/master/Clock/Config-Example.py#L14

Digital mode can be selected through the `digital` field, changing it to a `1`.
https://github.com/n0bel/PiClock/blob/master/Clock/Config-Example.py#L25

RyUp1, 2017-09-25:
Awesome thank you so much man!! I wish I could give you like a 5 star rating haha. Would I just save an image to my pi that I wanted to use? Like, where does the image come from?

dankolbrs, 2017-09-25:
Yup!

The image needs to be on your Raspberry Pi so that it can access and load to display on the clock. The easiest approach would be:
1. Copy your image into your `/home/pi/PiClock/images` directory
1. update the `background` field in `/home/pi/PiClock/Clock/Config.py` to `images/<NAME_OF_YOUR_IMAGE>`

If you're running the Pi under a different user/path anything like that, of course update the path as needed *if you don't know what I mean then you probably don't need to worry about it*.

From your comments, I'm not sure how familiar you are with Linux or the Raspberry Pi. If you need help getting the image on your Pi I'd recommend something like [WinSCP](https://winscp.net/eng/download.php), [Filezilla](https://filezilla-project.org/), or [MobaXterm](http://mobaxterm.mobatek.net/) to have a pretty easy interface to copy files to your Pi from your PC (assuming Windows). If it's from your phone, I'm a big fan of [ES File Explorer](https://play.google.com/store/apps/details?id=com.estrongs.android.pop) (assuming Android).

Please go ahead and update in here if you get it figured out or have another question :D

RyUp1, 2017-09-26:
Worked perfectly. You've been a great help. One last question for now though..how did you wire your pi to the monitors power supply? Thanks again!

dankolbrs, 2017-09-27:
I can't help you there, unfortunately, I've always powered it separately. The process would be using a multimeter to find a power output from the monitor's power supply to solder on to, use a step down regulator if it's not 5v (it'll probably be 12v), and wire up to either a USB output (if you've got one to chop up) or to the GPIO pins to provide power. 

You should be able to dig up a detailed walkthrough of these various steps with a few google searches, or someone might chime in here with more details.

n0bel, 2017-09-27:
@dankolbrs Thanks so much for responding.

@RyUp1 For most of my monitors, I just opened it up and poked around for its main power, which in most monitors is 12V or 5V.  I usually would choose the 12V because the 5V regulators usually don't supply much current.   Then I take the 12V down to 5V with a 3A or 5A buck converter, and run the Pi from that.   In other cases, the monitor has a USB connector which happened to supply enough current (luckily) to run the Pi with only the lighting bolt (or rainbow square) showing up during boot.

RyUp1, 2017-09-30:
Hello again!

So I have decided to take the next step and try to install some of the
optional things. Started with the thermometer. First, I saw that you
recommended using the ModMyPi guide to wire the sensor up. Is that the
easiest way, with a breadboard? Next, I tried to do step 8 which I pasted
below. After I typed that first part in (the first line, everything before
the python part) it would ask me for what I think was a github username and
password. I tried to just type my own in and it didn't work. I also tried
to just check out the code on github and attempt to install it via  pip
install w1thermsensor which installed something but did not get me any
further lol. I feel like I am learning but this stuff always blindsides me.
And I still have the remote and maybe LEDs to figure out next! Thanks for
your help again, Ryaner])

git clone https://github.com/timofurrer/w1thermsensor.git && cd w1thermsensor
 python setup.py install

dankolbrs, 2017-10-02:
The breadboard would be easiest to keep everything straight and ensuring you're hooked up correctly. If you make any mistakes, it's much easier to change around. If you've got room in your project, or just want to start with that to make sure everything works, that'd be the way to go. Once you feel good about it, you could move it from the breadboard and wire it up.

For the git clone not working, I would guess that there may have been a typo in the URL you used as the clone should work just as you cloned PiClock. Regardless, since you've already done the pip install you should be good as the latest release of `w1thermsensor` is `1.0.4` and that is what is on pip. No need to reinstall. You can verify by running `python -c "from w1thermsensor import W1ThermSensor"` which should output nothing. If you get what's below, it's either not installed or not available to that interpreter. 
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named w1thermsensor
```

closed 2018-12-23 by n0bel