n0bel/PiClock #90: app randomly crashes

derekzernach, 2017-10-13:
First off, thanks for a great app! I have it on a little touch screen by my door so I can quick check the weather before I leave.

It's been running great for months up until a couple weeks ago. It seems to randomly crash when i'm switching between the primary and secondary screens. I checked the logs and i'm getting Segmentation Faults....I saw you had issue with this back in 2015 and nothing since...

hsoj95, 2017-10-14:
I'm not on the development team, but thought I might jump in and offer a few possible solutions outside of the PiClock itself. First make sure your Pi is updated, as I know that the PiClock runs on several dependencies such as Python, QT, etc, and these need to be updated time to time. Also, maybe make sure you've updated the PiClock itself. You can look at the instructions for the clock to see how to do that. If neither of those solve your problem, try reinstalling the clock again. I know that's a pain as it means setting up the location and customizations you've made, but it might solve the issue. If those don't work then it most likely is purely an issue with the clock itself. If you can, you might consider pasting the part of the error log that shows the specific error, and any other info that might be relevant to that. That would probably help the development team find what is causing this issue.

I did notice I had a crash once or twice on the clock as well when switching screens, but I specifically noticed that happened after I did a round of updating on my Pi not too long ago, so I wonder if a core package the clock runs on got updated or not and is spawning errors.

Knackster, 2017-10-14:
Also, make sure the wall wart is 2.5 amps. The pi is going to be using all of this. Stack on HDMI out and wireless out and you just about reach the max current.

hsoj95, 2017-10-14:
@Knackster Oh absolutely! I didn't think about that being a cause of the segmentation fault, but I'd be wiling to bet that a power shortage would cause logic errors. I use a 3 amps because I overclocked mine, but a 3 amp power supply insures you have extra room to work with.

derekzernach, 2017-10-16:
@Knackster @hsoj95 Thanks for the suggestions, i'll take a look into both of those and see if that resolves the issue!

closed 2017-11-23 by n0bel