n0bel/PiClock #93: piclock not start

matteos1, 2017-12-05:
on reboot  piclock not start.. why?

dankolbrs, 2017-12-07:
We'll need some further information to help here. 

Assuming you're using cron with `@reboot` to start it, can you provide the output from 
```
crontab -l
```
That will show that it is configured in cron.

Next, depending on your OS/Hardware, etc, check the system logs to make sure it started on boot. For Rasbian or any other system, this would be
```
sudo journalctl --unit cron
```
We'd be looking for output such as
```
Dec 05 21:23:01 localhost systemd[1]: Started Regular background program processing daemon.
Dec 05 21:23:01 localhost cron[766]: (CRON) INFO (pidfile fd = 3)
Dec 05 21:23:02 localhost cron[766]: (CRON) INFO (Running @reboot jobs)
Dec 05 21:23:06 localhost CRON[827]: pam_unix(cron:session): session opened for user pi by (uid=0)
Dec 05 21:23:06 localhost CRON[865]: (pi) CMD (cd Projects/PiClock; sh startup.sh)
```

So those things would make sure its configured to start on reboot, and that it gets run on reboot.

To see if it's just crashing out on start it manually to see if there's any output there:
```
cd PiClock && sh startup.sh -n
```

If all else goes well, check the PyQtPiClock.log and provide any output from errors there. Be sure to sanitize the log and not do a direct copy/paste, as you want to make sure you don't post any of your personal API keys.

namedtoaster, 2018-01-07:
I'm having the same issue. My root user crontab is working (i.e. it's rebooting at the specified time), but the regular user crontab doesn't seem to want to run the @reboot command. The output of 
`sudo jornalctl --unit cron`
is similar to yours, with the addition of the following two lines:
`Jan 07 15:09:59 piclock CRON[335]: (CRON) info (No MTA installed, discarding output)
Jan 07 15:09:59 piclock CRON[335]: pam_unix(cron:session): session closed for user pi`
I don't know if that has any significance.

The only error I can find in the log when running that last command is the following:
`Traceback (most recent call last):
  File "PyQtPiClock.py", line 239, in wxfinished
    f = wxdata['hourly_forecast'][i * 3 + 2]
IndexError: list index out of range
get radar api call spacing oneshot get i=0.1`
but that doesn't seem to have anything to do with my original issue. I suppose it might

dankolbrs, 2018-01-24:
Not completely sure, but I don't think that the traceback given would result in PiClock not starting. I think it would still start, just have a failure in displaying radar.

It's not clear from the comment, does it start when running it manually? If it just doesn't start via the `@reboot` in cron, it'd probably be a path or permissions error. You could try to write the output to a log file with something like
```
@reboot cd PiClock && sh startup.sh 2>&1 >${HOME}/piclock_start.log
```

The MTA not installed provided by journalctl says that the output is not being sent via mail, so updating the cronjob as given above should write stdout and stderr to a file to get a better idea on what is going wrong. If after a reboot ${HOME}/piclock_start.log doesn't exist, check to make sure that the ${HOME}/PiClock directory exists and is read/writable by the `pi` user (assuming that is the user that has PiClock cloned and the cronjob to start it on boot).

closed 2018-12-13 by n0bel