n0bel/PiClock #94: Background slide show

bordeb, 2017-12-21:
Hy! First thumbs up and tnx for a great and long lasting open source project!
I have one issue. How to configure the slideshow - background?
I have found the authors comment https://hackaday.io/project/6184/instructions/discussion-87716 that it is possible and can be seen in closed issues here, but I have reread two times the closed  issues (as well as open ones) and did not found any answer nor clue how to achieve this.
I have tried also with transparent png as a background thinking that it might be combined with some other slide show image rotator in the background but this wasn't successful because the transparent png makes black background of the piclock. I would really much appreciate any help because this project is so interesting and this would be great added value.

bordeb, 2018-01-23:
Anything, just even a hint would be appreciated very very much! Thank you in advance!

dankolbrs, 2018-01-24:
Hi @bordeb, sorry there was no reply. Thanks for sticking with it and adding another comment.

I looked through the issues before, and didn't see this specifically referenced anywhere. I'm wondering if #47 is the one that came to mind in the comment you referenced. That issue has a link for a perl script for changing the config based on the weather (note, I didn't look at the script).

The approach I would suggest would be to set up a timer function to change the background on a set schedule. This would be setting up a timer such as is done [here](https://github.com/n0bel/PiClock/blob/master/Clock/PyQtPiClock.py#L355-L357), and in a newly created method change the background like is [done here](https://github.com/n0bel/PiClock/blob/master/Clock/PyQtPiClock.py#L817-L818). Basically it'd be similar to how weather/images are pulled and updated, but instead being selecting an image from a directory and updating the frame to display that image.

Please update this if that helps, or with other questions!

n0bel, 2018-01-24:
@dankolbrs Thanks for coming up with a direction for him to take.

bordeb, 2018-01-24:
@dankolbrs thank you very much on your kind help! I will dive into and see if I can get the results wanted and if I accomplish I will respond with the result or maybe additional questions if needed :-)
Thank you @n0bel also for noticing this, great job! congrats!

Njupko, 2018-11-20:
@bordeb did u get some results? I want to make same thing if u have done it. Thanks

bordeb, 2018-11-20:
@njupko, no I didn't unfortunately. Tried but then left aside bacause there was no time to learn all needed things to be able to sucesfully program it myself. If you are going to try I think that this could really be great added value.

Njupko, 2018-11-20:
@bordeb unfortunately i am not very skilled programmer but i am trying to make it work. I know how to define 1 picture but don't know how to make it read more then one from specific folder. If you can help me with that or @dankolbrs  and @n0bel  join in to make it work 😄

Njupko, 2018-11-22:
looks like i don't have enough knowledge to pull this off. If someone do this please replay

togatown, 2019-01-29:
> 
> 
> looks like i don't have enough knowledge to pull this off. If someone do this please replay

Edit your Config.py and add the following lines: (assumes your pictures are in 'images/slideshow')
```
background = ''
# SlideShow
slideTime = 5       # in seconds
slides = 'images/slideshow'
```
Set the background in Config.py to ''
```
background = ''
```
Locate the [styles for frame1](https://github.com/n0bel/PiClock/blob/3f94fd4c7322ee5350f52ce88fcfa004f3d92f21/Clock/PyQtPiClock.py#L992) and set the background color to whatever you prefer. [How?](https://www.w3schools.com/cssref/pr_background-color.asp)
```
frame1.setStyleSheet("#frame1 { background-color: black; border-image: url(" +
Config.background + ") 0 0 0 0 stretch stretch;}")
```
After [frames.append(frame1)](https://github.com/n0bel/PiClock/blob/3f94fd4c7322ee5350f52ce88fcfa004f3d92f21/Clock/PyQtPiClock.py#L994) and BEFORE any frame1 children (to be safe right after the line I posted) paste the following:
```
imgRect = QtCore.QRect(0, 0, width, height)
objImage1 = SS(frame1, imgRect, "image1")
```
Locate the [qtstart function](https://github.com/n0bel/PiClock/blob/3f94fd4c7322ee5350f52ce88fcfa004f3d92f21/Clock/PyQtPiClock.py#L419) and after [global objradar4](https://github.com/n0bel/PiClock/blob/3f94fd4c7322ee5350f52ce88fcfa004f3d92f21/Clock/PyQtPiClock.py#L425) add the line: (match the indents!)
```
global objImage1
```
after the [last line of the function](https://github.com/n0bel/PiClock/blob/3f94fd4c7322ee5350f52ce88fcfa004f3d92f21/Clock/PyQtPiClock.py#L449) add: (match the indents!)
```
objImage1.start(Config.slideTime)
```
Underneath this and before [class Radar](https://github.com/n0bel/PiClock/blob/3f94fd4c7322ee5350f52ce88fcfa004f3d92f21/Clock/PyQtPiClock.py#L452) add the following class to your file, 
```
class SS(QtGui.QLabel):
    def __init__(self, parent, rect, myname):
        self.myname = myname
        self.rect = rect
        QtGui.QLabel.__init__(self, parent)
        self.pause = False
        self.count = 0
        self.imgList = []
        self.getImageFiles(Config.slides)

        self.setObjectName("slideShow")
        self.setGeometry(rect)
        self.setStyleSheet("#slideShow { background-color: transparent; }")
        self.setAlignment(Qt.AlignHCenter | Qt.AlignCenter)

    def start(self, interval):
        self.timer = QtCore.QTimer()
        self.timer.timeout.connect(self.switchImage)
        self.timer.start(1000 * interval + random.uniform(1, 10))

    def stop(self):
        try:
            self.timer.stop()
            self.timer = None
        except Exception:
            pass

    def switchImage(self):
        if self.imgList:
            if not self.pause:
                if self.count == len(self.imgList): self.count = 0
                self.showImage(self.imgList[self.count])
                self.count += 1
                # if animFlag: count += 1
                # else: self._count -= 1

    def showImage(self, image):
        image = QtGui.QImage(image)
        bg = QtGui.QPixmap.fromImage(image)
        self.setPixmap(bg.scaled(
                self.size(),
                QtCore.Qt.KeepAspectRatio,
                QtCore.Qt.SmoothTransformation))

    def getImageFiles(self, path):
        try:
            dirContent = os.listdir(path)
        except OSError:
            print("path '%s' doesn't exists." % path)

        for each in dirContent:
            fullFile = os.path.join(path, each)
            if os.path.isfile(fullFile) and (fullFile.lower().endswith('png') or fullFile.lower().endswith('jpg')):
                self.imgList.append(fullFile)

    def playPause(self):
        if not self.pause:
            self.pause = True
        else:
            self.pause = False
```
Locate the [myquit function](https://github.com/n0bel/PiClock/blob/3f94fd4c7322ee5350f52ce88fcfa004f3d92f21/Clock/PyQtPiClock.py#L777) and edit [global objradar1, objradar2, objradar3, objradar4](https://github.com/n0bel/PiClock/blob/3f94fd4c7322ee5350f52ce88fcfa004f3d92f21/Clock/PyQtPiClock.py#L778) as follows:
```
global objradar1, objradar2, objradar3, objradar4, objImage1
```

Place the images, either jpg or png in the folder specified in the config and it should work. This is a preliminary build of this and I may modify it later. I'll post if I do.

n0bel, 2019-02-01:
@togatown Pull request?

togatown, 2019-02-03:
> 
> 
> @togatown Pull request?

Sure, I created a fork and made a Pull Request from it.

trcyprkr, 2022-01-12:
I haven't seen it so far, but has anyone asked or answered about auto scaling the background images while in slideshow mode?