n0bel/PiClock #95: Tons of network usage

ScottChapman, 2018-01-04:
I'm running a pretty old (maybe 1.5 years or so) version. And I noticed that the Pi that it runs on is using a ton of network bandwidth (like nearly 10GB down per day and 1.5GB up per day).

I haven't figured out if this is entirely due to PiClock, but that is the major thing running on it at the moment.

Is this normal?

dankolbrs, 2018-01-07:
I don't track network usage on specific devices to compare that to, but that is crazy high network use on that Pi. High enough that I would question what would even be running on the Pi to generate that much usage. 

If using the same values as the basic config example, radar refreshes every 10 minutes, weather every 30 minutes, and the map background happens on load. I did a `wget` on each one of the resource URLs to see approximately the size of the resource that is downloaded for each call to get a rough idea of expected network usage. 

Google map image for radar:
77K.  Two radar screens, 154K

Radar image gif from weather api
84K, again, double for two radar screens, 168k. Multiply 144 times for running every 10 minutes, 24192K

Weather data from weather API:
320K, Running 48 times a day, 15360K. 

Total comes up to 39706K, or around 40MB a day down using standard config. This is just a rough estimation as the image sizes probably vary, this assumes no screen switching or other interaction, and probably a few other things that I'm not thinking of. I would think that even if the settings are cranked up for faster radar and weather refreshes, rate limiting by the APIs would probably kick in way before 10GB a day down.

ScottChapman, 2018-01-11:
Yup, figured it out. false alarm...

closed 2018-01-11 by ScottChapman