n0bel/PiClock #97: dynamic gps updates

talitha77, 2018-01-20:
Love the clock.. would love to be able to update gps co-ords on the fly, as I'm on a ship..
this would be a great bridge clock.
J

talitha77, 2018-01-21:
i can save the lat/long to a file every hour say.. I already do for various other applications
maybe the clock could optionally retrieve the location from a file?

dankolbrs, 2018-01-24:
I love this issue. The use cases of PiClock are amazing. I feel pretty lame I've just got it running on a crappy laptop sitting on a shelf. Thanks again for creating this @n0bel.

I just wrote up a little in #94 about creating a timer task to update the background image. A similar approach could be done to read a file with updated GPS coordinates. 

I would also recommend taking a look at GPSD, especially if there are other applications that are already looking for updated GPS coordinates. I haven't interfaced with it directly, but have created a "fake" GPSD service for applications that use it. Depending on where the GPS information comes from and the applications that are consuming it, that may be a better method of integration than writing files as it would be reading TCP data. (Just a thought, I don't know your use case)

n0bel, 2018-01-24:
@dankolbrs You're way more creative than I am... I read this note and came up with nothing constructive to recommend.

talitha77, 2018-02-01:
I can update the config file with new gps co-ords but as the clock only loads the conf on startup it means a daily restart.. which isn't that bad :)

alphamobius, 2018-02-08:
Could the PyQtPiClock.py be changed to link to another python file/method that polls for the latest GPS coords off of a USB device?  GPSd seems like the best way to do this, but I don't know where to link this into PyQtPiClock.py (see lack of comments issue).

talitha77, 2018-02-11:
I really only need hourly updates. I've got a kml file from google earth that i am trying to use but so far no luck :(

dankolbrs, 2018-02-14:
My previous comment linked up the issue where I provided info on how to change the background. This isn't much different.
1. Create method to update location
1. Create timer object calling that method

The method to update location can be another object, reading a file, reading from GPSd, etc. However you want to get the lat and long, it would then need to update the Config.wulocation data and radar markers.

Providing a more specific example than previously, getting the temperature from the sensor on the pi is set up this same way. 
1. [Method that gets temperature](https://github.com/n0bel/PiClock/blob/master/Clock/PyQtPiClock.py#L155-L163)
1. [Timer object](https://github.com/n0bel/PiClock/blob/master/Clock/PyQtPiClock.py#L364-L366)

I'd love to see more contributions, and this would be a good one for someone to get to know the internals with.

talitha77, 2018-02-15:
I've managed to get GPSd to read from a UDP source, so I have the lat and long on the PI now but thats where i'm getting stuck. I'm pretty low grade at linux.. I'll soldier on though. Thanks for the tips :)

talitha77, 2018-02-23:
I've cracked it..  magic stuff.. thanks all :)

merky1, 2018-03-07:
I use an external perl script on a cron schedule to change the background colors based on temperature.  The downside of this is having a large pool of config files for each GPS location I am at when I travel with the clock.  It would be nice if there was a way to put "universal" settings (gps, timezone) in a known file separate from creature comforts like image.  Something akin to the way API keys are handled.  This would allow for dynamic updating without reloading entire configs.