n0bel/PiClock #101: LED light problems - NeoAmbi.py run issue...

cranesoft, 2018-04-14:
First, thanks to all for the effort put into this awesome project!  Love the clock!

I've had my first PiClock up and running for a few days now and wanted to add lights.  startup.sh isn't able to run NeoAmbi.py.  After an attempt to run NeoAmbi.py directly I think I see the issue:
`pi@redfox:~/PiClock/Leds $ python NeoAmbi.py
Traceback (most recent call last):
  File "NeoAmbi.py", line 3, in <module>
    from NeoPixel import NeoPixel, Color
ImportError: libboost_python-py27.so.1.49.0: cannot open shared object file: No such file or directory
`
During my install of PiClock I wasn't able to find libboost-python1.49.0 (apt-get install libboost-python1.49.0) so installed what I found: apt-get install libboost-python1.62.0

Going back to my problem, it looks like the shared object file PiClock/Leds/NeoPixel.so was built for libboost-python1.49.0 not *1.62.0.  I haven't had any luck finding info on rebuilding NeoPixel.so.

I'm running this on the new Rasberry Pi 3 Model B+ using Raspbian.

Any pointers would be much appreciated!
Cheers!

n0bel, 2018-12-13:
all the NeoPixel code for python is based on this:
https://github.com/jgarff/rpi_ws281x/tree/master/python

n0bel, 2018-12-23:
I'm guessing this is a Stretch issue, which I'll deal with when I redo the instructions for Stretch

togatown, 2018-12-25:
According to the link n0bel posted, the old code is deprecated. 
To get and install the newest version run: 
sudo apt-get install scons swig
git clone https://github.com/jgarff/rpi_ws281x
cd rpi_ws281x
sudo scons
cd python
sudo python setup.py build
sudo python setup.py install

In you PiClock folder, edit startup.sh
Replace: python -c "import NeoPixel" >/dev/null 2>&1
With: python -c "import neopixel" >/dev/null 2>&1

Replace NeoAmbi.py with the following
```
import time
from neopixel import *
import argparse

#LED strip configuration:
LED_COUNT      = 145     # Number of LED pixels.
LED_PIN        = 18      # GPIO pin connected to the pixels (18 uses PWM!).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 10      # DMA channel to use for generating signal (try 10)
LED_BRIGHTNESS = 60     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL    = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53

# Define functions which animate LEDs in various ways.
def colorWipe(strip, color, wait_ms=50):
    """Wipe color across display a pixel at a time."""
    for i in range(strip.numPixels()):
        strip.setPixelColor(i, color)
        strip.show()
        time.sleep(wait_ms/1000.0)

def wheel(pos):
    """Generate rainbow colors across 0-255 positions."""
    if pos < 85:
        return Color(pos * 3, 255 - pos * 3, 0)
    elif pos < 170:
        pos -= 85
        return Color(255 - pos * 3, 0, pos * 3)
    else:
        pos -= 170
        return Color(0, pos * 3, 255 - pos * 3)

def rainbowCycle(strip, wait_ms=20, iterations=5):
    """Draw rainbow that uniformly distributes itself across all pixels."""
    for j in range(256*iterations):
        for i in range(strip.numPixels()):
            strip.setPixelColor(i, wheel((int(i * 256 / strip.numPixels()) + j) & 255))
        strip.show()
        time.sleep(wait_ms/1000.0)

# Main program logic follows:
if __name__ == '__main__':
    # Process arguments
    parser = argparse.ArgumentParser()
    parser.add_argument('-c', '--clear', action='store_true', help='clear the display on exit')
    args = parser.parse_args()

    # Create NeoPixel object with appropriate configuration.
    strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)
    # Intialize the library (must be called once before other functions).
    strip.begin()

    print ('Press Ctrl-C to quit.')
    if not args.clear:
        print('Use "-c" argument to clear LEDs on exit')

    try:
        while True:
            rainbowCycle(strip)
    except KeyboardInterrupt:
        if args.clear:
            colorWipe(strip, Color(0,0,0), 10)
```

n0bel, 2018-12-26:
@togatown No need to do all of that.. 
The instructions are updated. 
NeoAmbi.py is a circular rainbow that turns based on time.

Library install:  
sudo pip install rpi_ws281x

NeoAmbi.py is also updated.

startup.sh updated (wasn't before but is now, thanks @togatown )

update.py updated to inform people of the new library if it doesn't exist.

togatown, 2018-12-26:
@n0bel Thanks, the code I posted was a circular rainbow as well. I looked at the NeoAmbi.py updates and was playing with the code, interrupt 5 is locking up my Pi pretty consistently.

I am using MQTT to control my lights but I'm sure I can modify your code like I did the sample snippet I posted. I plan on being able to change patterns or choose solid single colors.

I'm really not impressed with the LED performance on the Pi.  They are super fast on the Arduino and that's 8 bit. I am using the fastLED library on the Arduino but it shouldn't make that much difference.

I may just move my lights to an ESP8266, ESP32 or D1 Mini.

ScottChapman, 2018-12-26:
Been following this, sorry but I don't know what you guys are talking about? Circular Rainbow? I don't know what it is, but I think I want that! ;-)

n0bel, 2018-12-26:
@ScottChapman In the following picture, notice the colors behind the clock.   This is done with a neopixel strip behind the clock.   The rainbow rotates in time with a 1 minute progression.

[image] 

Some details about it appear on this image:
[image]

n0bel, 2018-12-26:
@togatown I have some Pi's running both NeoPixel and DotStar type led strips.   The longest is 450 pixels, and I've not run into any performance issues (esp on a Pi3).  Things break down more because of the 800khz limit of the NeoPixel.   DotStars I clock at 30Mhz on the 450 pixel string.   In this application it is designed to be a slow rotation. The fact that the clock takes about 60% cpu on a Pi Zero and about 25% on a Pi3, I saw no reason to bog down the pi with more computations.

n0bel, 2019-01-12:
New Neopixel software is in the latest install instructions and Leds folder.

closed 2019-01-12 by n0bel