n0bel/PiClock #106: Map data not loading

mkbcaptain, 2018-10-13:
Lately I am having the issues of the google map not loading. Radar and weather loads. But maps are not loading. Suggestions?

desolat68, 2018-10-17:
It looks like they're giving you a 300 dollar credit for 12 months. So as little as we will pull with the PiClock, we won't use the 300 dollars. However, we WILL use the 12 months up of course, at which time you have to bill your card for another chunk of time. In other words, even if you follow the above link and sign up for the free trial, the maps on the PiClock will be blank eventually. I wonder if PiClock 2 will have this issue......

Cooperstown39, 2018-10-17:
> It looks like they're giving you a 300 dollar credit for 12 months. So as little as we will pull with the PiClock, we won't use the 300 dollars. However, we WILL use the 12 months up of course, at [...]

Assuming I read it correctly, the WU api will expire for everyone on 12/31/18.  I was just trying to get to the end of the year in hopes PiClock2 will be out by then.

n0bel, 2018-12-13:
@Cooperstown39 https://cloud.google.com/maps-platform/pricing/sheet/ As I read the google maps api pricing page, you get $200usd credit per month forever, not just 12 months.  The map is only loaded once per start of the clock, so even if you have your PiClock reboot every day, you end up with max of 31 map calls per month, at a rate of $2.00 per thousand, which is $0.062 per month.

n0bel, 2018-12-13:
@vanhalen26505 Everything seems to work fine with my google api keys, no grey maps.   Google did recently stop giving away maps (depending where your IP originates) for free, so things turned grey.  Just do the api thing.   I'll be updating the instructions to remove the wording about removing your googleapi key.

vanhalen26505, 2018-12-13:
OK, I will give it another shot from scratch. I had modified the source
code a bit. Perhaps I goofed up something additional. Thanks very much for
your time and input. Much appreciated! The clock appears on a bunch of
panels in my house and makes it the coolest house (my opinion) on the
planet! :)

vanhalen26505, 2018-12-14:
[image]
I am thrilled to say it is back up and working! :) Not sure if I had to enable more things on the Google side with the API key, but it starting working again tonight! Thanks!!

vanhalen26505, 2018-12-14:
Also, please excuse my ignorance, as I am guessing there is a better place to share things like these pics. If you can tell me where, and think others might like them, I am happy to recreate them in another location. 

I took an analog out from my Pi and fed it into a channel on my security DVR. Did the same with a couple other PIs that I have monitoring four traffic cams each. The rotation of screens happens every 20 seconds, so I get PiClock, first set of four cams, and second set of four cams mixed in with 19 cameras showing my property. When motion is detected on any of the 19, it goes full screen. 

These pictures show the 60" panel above my fireplace in our great room. I also have it on a number of other panels around the house, including one between the showers in our master bath, so we can see traffic and weather while getting ready in the morning. LOVE IT!!! Thanks so much again for given us this great piece of code and project. It really help make my house the coolest one I know, and friends and guests marvel at it!

[image]
[image]
[image]
[image]

vanhalen26505, 2018-12-24:
> @vanhalen26505 <https://github.com/vanhalen26505> Thanks for the great
> pictures of your build!

You bet! Thanks for the excellent code and making all this happen for us.
So appreciated!!

vanhalen26505, 2019-01-14:
> I would like to learn more about how you incorporated the traffic cams
> with your PiClock. Care to share that?

Sure thing. To be clear, they are not actually part of the PiClock. I have
three Pi computers running, and each of them us going from an analog out to
a Hikvision security camera DVR. The DVR is connecting to a bunch of
screens in my house through an HDMI multiplier system, so each TV gets the
feed.

One of the Pi's is running PiClock. The other two are each running four
traffic cameras. The DVR shows sets of cameras in a pre-determined sequence
and rotates them about every 20 seconds. The two Pi's that are running the
four traffic cameras each are connecting to RTMP (I think that is the
protocol) feeds.

My state has a bunch of publically-accessible traffic cams. I found a
little bit of code that plays RTMP feeds on github, I believe. The person
who posted it had made a security camera system with Pi's for a pizza shop,
so that the owner could be making food in the kitchen and watching the rest
of the place on a monitor.

I basically did a light adaption of that with the IP addresses for the
public traffic cams.

It might be possible somehow to modify the Python code from PiClock to
rotate to show RTMP feeds periodically, but, I will admit, I am not that
good of a coder! :)

Hope this info helps. It is pretty cool to be able to glance and see the
weather and traffic when getting ready in the morning. Someday, I want to
try to get some news headlines mixed into the feed as well. Dreams...

Cheers,
Lee