n0bel/PiClock #108: Day forecast test truncation

ScottChapman, 2018-12-21:
Many of the day forecast text appears to be truncated. It would be really good to have the text wrap.

I realize this would likely require a fairly UI change. Perhaps you could do:
```
------------------------------
| Thursday            70% 54/32 |
| Mostly cloudy with periods  |
| of rain throughout day.         |
------------------------------
```

That way it is still 3-lines of text per day

n0bel, 2018-12-21:
I've seen it, but I haven't decided what the right answer is.   It didn't seem (to me) that the additional information was more useful.    For me I find the numeric version of the forecast more meaningful, since it shows Rain or Snow % and amounts.    So things won't always fit on 3 lines either.   A solution may be to make the whole box one big wrapped text box.    Then we loose the nice formatting.    Notice my forecast for Wednesday, it won't fit on 3 lines either:
[image]

n0bel, 2018-12-23:
Formatting changed on commit:  f191cf7b8d13123392cfec66e75026b98484c0af
Its still not ideal, but I think the best at this point.

ScottChapman, 2018-12-31:
Yea, looks better. But there is some overlap now. Not sure, but perhaps you can reduce the gap between the top three lines?
[image]

n0bel, 2019-01-12:
nomatter what choice is made, there will be overlap, or the text will be too small for the short things.
A font size change may be the next choice.

closed 2019-01-12 by n0bel